prusa3d/PrusaSlicer #5846: Ender 3 Printer presets (and likely others) hold nozzle warm whilst waiting for bed

strayr, 2021-01-22:
### Version
2.3.0 but likely version indepedent
### Operating system type + version
win 10 64 bit

### 3D printer brand / version + firmware version (if known)
Ender 3v2, marlin 2.0.7.2
BUT: probably applies to most profiles

### Behavior
* _Describe the problem_
Bed takes ~5 mins to heat to indicated 80C, hotend reaches ~220C in <2mins. Hotend oozes whilst waiting for bed. Turns into a blob on the nozzle that catches somewhere in the first layer if i don't catch it as printing starts.
Whilst the behaviour reduces time to print, it increases likelyhood of failure and blemishes.

* _Steps needed to reproduce the problem_
generate g-code with supplied profile, printing optional

* _Expected Results_
Bed reaches indicated temperature, hotend starts heating, no ooze.

>M140 S[first_layer_bed_temperature] ; set bed temp
>M190 S[first_layer_bed_temperature] ; wait for bed temp
>M104 S[first_layer_temperature] ; set extruder temp
>M109 S[first_layer_temperature] ; wait for extruder temp

Even better: hotend starts heating when bed it 80% to temperature (is this possible?) Feature request?
Maybe there is a way to do this in the macro syntax?

>M140 S[first_layer_bed_temperature] ; set bed temp
>M190 S[first_layer_bed_temperature * 0.8] ; ****wait for 80% of bed temp****
>M104 S[first_layer_temperature] ; set extruder temp
>M190 S[first_layer_bed_temperature] ; wait for bed temp
>M109 S[first_layer_temperature] ; wait for extruder temp

* _Actual Results_

>M104 S[first_layer_temperature] ; set extruder temp
>M140 S[first_layer_bed_temperature] ; set bed temp
>M190 S[first_layer_bed_temperature] ; wait for bed temp
>M109 S[first_layer_temperature] ; wait for extruder temp

Blob of filament drool sometimes caught by intro line, sometimes ends up in middle of first layer.

_Is this a new feature request?_
Partly, see even better above, it's mostly a suggested change in default behaviour

rtyr, 2021-01-22:
I personally prefer current behavior and I consider it as "common standard". Same sequence is used in Ender3 profile bundled in the Cura. I will leave it to the @pmjdebruijn, who added most of the Creality printers. What do you think?

> hotend starts heating when bed it 80% to temperature (is this possible?)
No it is not possible.

strayr, 2021-01-22:
I'm seeing the common standard as a mild bug that's likely to cause more problems as users try to find gcode to alert them of the exact moment the bed reaches temperature and printing starts and get in there with a popsicle stick. Slight risk of injury and slight quality issue, and some gcode hacking for experienced users. To me it feels like the standard is done this way becasue it was easy not because it's always the right thing to do.

pmjdebruijn, 2021-01-22:
I'll try to take a look over the weekend.

foreachthing, 2021-01-22:
> > hotend starts heating when bed it 80% to temperature (is this possible?)
> 
> No it is not possible.

Yes, it is possible.
[Here](https://help.prusa3d.com/en/article/macros_1775) (Section 3, Arithmetics _(btw: the * is missing between - and /)_), it says `M104 S{first_layer_temperature[0] * 2/3}`, so if I write `M190 S{first_layer_bed_temperature[0] * 0.8}`, I get exactly that: `M190 S48`.

rtyr, 2021-01-22:
I misunderstood what he meant. Somehow I understood it as he expects gcode to get feedback from the printer, my bad :-).

strayr, 2021-01-22:
> Yes, it is possible.
> I've been doing this `M190 S{first_layer_bed_temperature[0] - 5}` for a long time now...

That's really helpful, thankyou, I'll be throwing something like this into my starting gcode.

pmjdebruijn, 2021-01-23:
BTW, if oozing turns into a blob on your nozzle, I'm guessing you have a semi-clogged nozzle. Any oozing should just come out in a straight line, as if was extruding really slowly.

As as sidenote, what are you printing? 220C/80C seems like an odd combination, given that both are unusually hot for plain PLAs, but too cold for many other common filament types.

And I've tested this:

```
M190 S[first_layer_bed_temperature] ; wait for bed temp
M104 S[first_layer_temperature] ; set extruder temp
M109 S[first_layer_temperature] ; wait for extruder temp
```

And I still have oozing with common PLA (at 200C), it's just less, which is actually worse practically speaking, having a fair bit of oozing, makes the filament dangling off the nozzle much more convenient, and slightly safer, since you can grab the filament extrusion at the bottom further away from the nozzle to remove it in it's entirety, just before the priming line starts.

strayr, 2021-01-23:
> ...I'm guessing you have a semi-clogged nozzle. 
Nozzle is ok, it's printing consistently. The intro code doesn't home and park before heating, so the head's been sat quite low to the bed as printing a lot of small stuff. Nozzle drops onto the little spiral on the bed and makes a small blobby string. I'm used to the behavior of my Flashforge that oozes outside the print area, but that does a full travel from max-z to first layer and usally knocks off the ooze itself. But theres no heated bed. For all the little FF is annoyingly proprietory, it requires less babysitting.
 
> As as sidenote, what are you printing? 220C/80C seems like an odd combination

Ankun PLA+, recommended bed 70-80 nozzle 210-235. I can probably drop the bed temp a little, parts are coming lose from the (stock V2 cloated glass) bed between 40 and 50C with an audible ping. It's really tough compared to PLA but ooze strings don't break off.

pmjdebruijn, 2021-01-23:
The fact that you're not having immediate print issues, doesn't mean your nozzle is 100% fine though. Your filament should really ooze straight out of the nozzle, where you can just pull it away with little fuss. And with the current start gcode prints start fairly fast, so you won't even have to wait that long, as far as "babysitting" goes. So practically speaking I think it's a bit of a non-issue.

As far as "knocking off" any ooze, I'm not sure how reliable we can get that across all Creality printers and different filament types. And I'd prefer to keep behavior reasonably consistent between printers.

Having said that, I'm not convinced that changing the start gcode will be a clear benefit for most users at this point.

strayr, 2021-01-23:
Ok, a more simple fix and one that really needs to be done is the current start code looks like this

> G90 ; use absolute coordinates
> M83 ; extruder relative mode
> M104 S[first_layer_temperature] ; set extruder temp
> M140 S[first_layer_bed_temperature] ; set bed temp
> M190 S[first_layer_bed_temperature] ; wait for bed temp
> M109 S[first_layer_temperature] ; wait for extruder temp
> G28 ; home all

the first movement is the G28, it didn;t take me long to home and lift Z 100mm before turning the heaters on. Idiot profing i guess. But can't trust position of the nozzles beofre a move so need to home first.

Maybe i'm looking at this too much like an FFF printer is a product and should work with the least user intervention and least amount of user procedure when the developing community are makers and hobbyists.

I get that I'm going a bit further as my ender is semi-enclosed, there's just something I don't like about normal operating procedure being "stick your hand in the mechanism"

pmjdebruijn, 2021-01-23:
I think you'll like the changes in #5863, please try, and report back so @rtyr can merge...

strayr, 2021-01-24:
> I think you'll like the changes in #5863, please try, and report back so @rtyr can merge...

YES, that works for me and is behaviour I prefer. Thanks.

I still prefer to check Z-height before heating, code it better than relying on users not to be lazy. I don't know enough about ABL to know if it's possible to do (pseudocode for abstraction)

> // Start of instructions
> home z on one point
> lift Z
> wait for bed temp
> home and check mesh
> print object

But that's vastly preferable to 
> // Start of instructions
>heat on // Z is in unknown position
> Wait for temperature
> home procedure and level if printer supports it
> print

pmjdebruijn, 2021-01-24:
The main issue with your suggestion is that it would slow down print start times noticeably. Which begs the question if we should make everybody suffer this slowdown, to be able to catch a fairly extreme corner case. So no, it's not vastly preferable.

As far as ABL goes, we have two variants, a fast one for printers with a small bed (Ender-3) and a slow one for printers with a larger bed (CR-10), as those are more prone to being warped. While the latter is technically more correct, the former halves print start time, with little or no tangible downsides in most cases.

pmjdebruijn, 2021-01-25:
There's also a counter argument to be made, that a nozzle should be heated so it does not damage the bed surface, if it has cold filament protruding during Z homing.

@rtyr as far as I'm concerned this issue is resolved.

rtyr, 2021-01-25:
Thank you. I will look into that pull request, hopefully soon.

closed 2021-01-25 by rtyr