prusa3d/PrusaSlicer #5852: 2.3.0 chrome linex no such file or directly

ShaneTheCreator, 2021-01-23:
2.3.0 Version
_Version of PrusaSlicer used goes here_

Chromebook/ Linux

PrusaSlicer-2.3.0+linux-64-202101111322.AppImage file saved to Linux files

In Linux Terminal:

LS shows PrusaSlicer-2.3.0+linux-64-202101111322.AppImage file

Terminal

/PrusaSlicer-2.3.0+linux-64-202101111322.AppImage

Will not run

"No such file or directly"

MasterCATZ, 2021-01-26:
I think they changed the profile folder location

bubnikv, 2021-03-12:
I think you are missing libglu

https://forum.prusaprinters.org/forum/prusaslicer/how-to-install-and-run-prusaslicer-on-chromebook-with-linux-support/

bubnikv, 2021-09-22:
No response. PrusaSlicer 2.4.0-alpha1 does not require libglu dependency, thus it shall run out of the box on Chrome OS.

closed 2021-09-22 by bubnikv