prusa3d/PrusaSlicer #5854: PrusaSlicer 2.3.0 slow on macOS

BastR, 2021-01-23:
### Version
2.3.0+

### Operating system type + version
macOs Big Sur 11.2 (20D53) (was already present in previous builds)
Mac mini M1 8go with LG27uk650 4k interface scaling with resolution 2560x1440

### 3D printer brand / version + firmware version
MK3S v3.9.2

### Behavior
PS becomes excessively slow after cutting if I enable support for retinal screens. If I disable retina-support, PS responds correctly. The problem may not appear on "simple" models, as soon as the model becomes a bit complex, the slowdowns are present and can lead to application freezes. I tried to wait to see if the situation would improve but no, I had to force stop the app.

At the moment I have deactivated the retinal support to be able to use the software.

I haven't done any more tests, such as changing the screen resolution, if necessary I can do it.
I don't know if it's specific to Mac M1 or to macOS in general.

[image]

homan85, 2021-01-24:
Same issue. Takes forever to load slice in MacOS 11.1, PS 2.3.0, when retina resolution is activated.

treyd3, 2021-02-10:
This is the same issue as [issue 5412](https://github.com/prusa3d/PrusaSlicer/issues/5412).
Given a sliced model with enough segments, the app becomes unresponsive in preview whether 'use retina resolution' is enabled or not.

bubnikv, 2021-02-11:
It will be fixed in the next release. Closing.

closed 2021-02-11 by bubnikv

bubnikv, 2021-04-14:
Fixed in PrusaSlicer 2.3.1-rc