prusa3d/PrusaSlicer #5855: Esun Filament

DrakeinRaven, 2021-01-23:
Hi all

where do i get support for eSun filaments in prusa? Im using PLA+ & PETG. 

Thanks

rtyr, 2021-01-23:
There are bundled profiles for eSun PLA and eSun ABS . Those are available for our printers and can be enabled via configuration wizard (Configuration - Configuration Wizard). If you need filament profile which is not yet bundled in the PrusaSlicer, you have to create it by yourself or use some generic profile (for example Generic PLA).

closed 2021-01-25 by rtyr

amatulic, 2022-02-28:
@rtyr - This should be re-opened. The OP specifically asked about PLA+, which is not the same as PLA, and there is no profile available for PLA+ bundled with PrusaSlicer.

As far as I can tell, the differences are primarily in temperature and retraction speed (slower than usual, like 25). Bed temperature also can be as low as 50°.