prusa3d/PrusaSlicer #5856: PrusaSlicer 2.3.0 update problem

Karol01pl, 2021-01-23:
Hello, I have a problem with PrusaSlicer 2.3.0 after installing the latest version, the slicer tells me to update. When I update with a slicer, the beta version is installed, what could be the problem?

Karol01pl, 2021-01-23:
@josefprusa

rtyr, 2021-01-23:
I am not sure what do you mean. Please provide more detailed info or at least some screenshots.

> after installing the latest version, the slicer tells me to update

This notification will pop-up only if you are running older version of the PrusaSlicer (for example alpha, beta, rc). It is also possible, that PrusaSlicer notifies you about the configuration update (it will only update bundled profiles, not the PrusaSlicer application itself).

PrusaSlicer 2.3.0 (final release) is available for download here https://www.prusa3d.cz/prusaslicer/ or here https://github.com/prusa3d/PrusaSlicer/releases.

Karol01pl, 2021-01-23:
> I am not sure what do you mean. Please provide more detailed info or at least some screenshots.
>
>
> This notification will pop-up only if you are running older version of the PrusaSlicer (for example [...]

Sorry I am using google translator. I have the latest version from the Prusa site installed and the slicer still tells me to update and when I do, I install the beta version

rtyr, 2021-01-23:
What operating system are you using?

> I have the latest version from the Prusa site installed.

What is the exact name of the file you are installing?

Please provide some screenshots.

Karol01pl, 2021-01-23:
https://imgur.com/a/1EYZpJW

Windows 10 Home 

PrusaSlicer_Win_standalone_2.3.0.exe

rtyr, 2021-01-23:
@Janys Can you please look into this?

Janys, 2021-04-19:
Hello, we have just released a new version 2.3.1. The updater should detect it correctly.

In case the issue persists, can you please try going to C:\Users\Public\Public Documents\Prusa3Dupdater\Updates and delete the update.aiu file? It may contain outdated information and this is what may cause issues with detection.

closed 2026-09-29 by prusaslicer-bot[bot]