prusa3d/PrusaSlicer #5858: Ironing on two levels with filament change incorrectly prompts for an second filament change

OakleyA, 2021-01-23:
### Version
2.3.0+win64

### Operating system type + version
Windows 10 Pro 20H2

### 3D printer brand / version + firmware version (if known)
original Prusa MK2.5S
Firmware 3.9.1

### Behavior
I'm printing a sign in two colours. I want the backing to be ironed, then a colour change for the text, then the text to be ironed.
The first ironing stage and filament change happens perfectly. When it's time for the final ironing stage, the printer prompts me to change the filament again when I didn't want it to.

#### Project File (.3MF) where problem occurs

[dont_touch_one_piece.zip](https://github.com/prusa3d/PrusaSlicer/files/5860533/dont_touch_one_piece.zip)

rtyr, 2021-01-26:
I checked your attached 3MF project. There is no ironing enabled, no filament change for the text. There is only one filament change added to the last layer (probably missclick/mistake?).

When I do the steps you described (and remove the unwanted color change at the last layer), everything is OK. I don't see any issue, gcode is also OK.

Final 3MF:
[dont_touch_one_piece_fixed.zip](https://github.com/prusa3d/PrusaSlicer/files/5870316/dont_touch_one_piece_fixed.zip)

rtyr, 2021-01-26:
Can we close the issue?

OakleyA, 2021-01-27:
Hi, apologies. That's not the original 3mf file - I tried to recreate the original and obviously failed. Please find attached the gcode which I still have.

Is it possible to accidentally add a filament change between the last layer and the ironing?

If you still believe this is user error after examing the attached gcode then please feel free to close this issue.
[dont_touch_one_piece_0.2mm_PETG_MK2.5S_2h1m.zip](https://github.com/prusa3d/PrusaSlicer/files/5882155/dont_touch_one_piece_0.2mm_PETG_MK2.5S_2h1m.zip)

rtyr, 2021-01-27:
I checked your gcode. There are two color changes inserted. One is inserted to the beginning of the last layer (same as in the 3MF you posted earlier). It is not between the last layer and the ironing. I still believe it was added accidentally by clicking on the "+" button. It is easy to overlook that on the last layer.

I am closing this issue.

closed 2021-01-27 by rtyr