prusa3d/PrusaSlicer #5859: PrusaSlicer 2.3 AppImage Crash on Slice

gooba42, 2021-01-23:
### Version
Version 2.3.0+linux-x64

### Operating system type + version
Kubuntu 20.10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S - FW next-to-most-recent but not involved

### Behavior
Slicing any model with any settings (i.e. Benchy with default PLA profile) will do most of the job up until the final stages where the GCode is exportable but the time hasn't been calculated yet. 

At that stage, instead of finishing normally PrusaSlicer closes itself without any reported errors. It leaves me nothing to screenshot and the issue happens well before I try to print.

#### Project File (.3MF) where problem occurs
[3DBenchy.zip](https://github.com/prusa3d/PrusaSlicer/files/5860548/3DBenchy.zip)

fkrecmer, 2021-01-23:
I have the same problem on Ubuntu 20.10 Cinnamon (Linux ryzenshine 5.8.0-40-generic #45-Ubuntu SMP Fri Jan 15 11:05:36 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux). 
Attaching debug and trace ouputs. Slicer crashes with "Segmentation fault (core dumped)". 

dmesg:
```
[ 1088.511846] slic3r_main[6100]: segfault at f39ccf0 ip 00007fefb28d3ff5 sp 00007ffcdaef8320 error 6 in radeonsi_dri.so[7fefb1e07000+dba000]
[ 1153.654487] slic3r_main[6310]: segfault at e27b0b0 ip 00007fd046330ff5 sp 00007ffe306b1820 error 6 in radeonsi_dri.so[7fd045864000+dba000]

```
journalctl:
```
led 23 21:14:46 ryzenshine kernel: slic3r_main[7061]: segfault at f1a3370 ip 00007f4ca34dcff5 sp 00007fff2f3baf90 error 6 in radeonsi_dri.so[7f4ca2a10000+dba000]
led 23 21:14:46 ryzenshine kernel: Code: 3d 00 03 00 00 0f 87 4a 01 00 00 48 be 1e ab a1 5c 03 00 0c 00 8b 42 0c 8d 48 03 89 4a 0c 48 89 c1 48 c1 e1 04 48 8d 44 0a 28 <48> 89 74 0a 20 44 88 28 44 88 60 01 48 85 db 74 52 48 8b 13 48 c7
```

Reproducible everytime (also with GTK3 appimage). Version 2.2 and below is working fine. 

[slicer-debug.txt](https://github.com/prusa3d/PrusaSlicer/files/5860970/slicer-debug.txt)
[slicer-trace.txt](https://github.com/prusa3d/PrusaSlicer/files/5860972/slicer-trace.txt)

lukasmatena, 2021-01-25:
This seems related to https://github.com/prusa3d/PrusaSlicer/issues/5859.

gooba42, 2021-01-25:
I can't speak for anybody else but in my case it's not multiple instances or anything "fancy" like that and I've not found success with anything.

tullo-x86, 2021-01-27:
Hi, I'm the reporter of #5889. I found a workaround on my machine, which was to disable Mesa's use of DRI3 by starting PrusaSlicer with `LIBGL_DRI3_DISABLE=true prusa-slicer`. Not sure if it'll work here, but it's worth a shot.

If it does successfully work around the issue, please say so; it could help the devs get to the bottom of it.

gooba42, 2021-01-30:
That appears to have fixed or at least sidestepped the issue for me. 

A side note - literally everything seems to also run far faster with this switch.

MasterCATZ, 2021-03-05:
LIBGL_DRI3_DISABLE=true ./prusa-slicer

this seemed to still seg fault for me 

simple small things like this keep seg faulting 
[P047_Belt.zip](https://github.com/prusa3d/PrusaSlicer/files/6089595/P047_Belt.zip)

LIBGL_DRI3_DISABLE=true ./PrusaSlicer-2.3.0+linux-x64-202101111322.AppImage 
will show the final slice then seg fault

tullo-x86, 2021-04-02:
@MasterCATZ what driver were you using for the R9 290?

lukasmatena, 2023-02-08:
Do you still have these issues with [2.6.0-alpha3](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.0-alpha3)?

gooba42, 2023-02-19:
I'm sorry, someone else might have to speak to this fix. My desktop machine died since this was filed so I don't have my Linux machine to test on. I can call this Closed - Can't Repro unless someone else thinks it needs to be reopened.

closed 2023-02-19 by gooba42