prusa3d/PrusaSlicer #5860: FR: Anti-aliased top layers

Battlevaria, 2021-01-23:
### Version
2.3.0

### Operating system type + version
macOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s

### Behavior
On a slanted top surface, the well-known stair-stepping problem ought to be fixable with anti-aliasing. For example, when printing the avocado boat (https://www.thingiverse.com/thing:2769468), the stair steps are clearly visible. Using the variable layer height feature, I was able to improve this model a fair amount, at the cost of having a lot of thing layers in most of the part. See photo for comparison (smaller layers on the right). 
[image]

This is a hack, though. It ought to be possible to make reduced stair-stepping a feature of PrusaSlicer: Interpolate the heights between "stair steps" and print thinner final layers in between. I expect printing the thinner layers first would work best, avoiding bringing a hot nozzle down close to a higher layer.

Without:

```
--------------    <-- 0.2mm layer
|            |
|            |
|            |
|            |
---------------------------------------
|                                     |
|                                     |
|                                     |
|                                     |
----------------------------------------------------------------
```

With:

```
--------------   <-- 0.2mm layer
|            |-----    <-- 0.16mm layer
|            |    |-----  <-- 0.12mm layer
|            |    |    |-----  <-- 0.08mm layer
|            |    |    |    |-----  <-- 0.04mm layer
---------------------------------------  <-- 0.2mm layer
|                                     |-----    <-- 0.16mm layer
|                                     |    |-----  <-- 0.12mm layer
|                                     |    |    |-----  <-- 0.08mm layer
|                                     |    |    |    |-----  <-- 0.04mm layer
-----------------------------------------------------------------
```

_Is this a new feature request?_
Yes.

foreachthing, 2021-01-23:
Is it maybe this your looking for: "Variable-Layer-Height"
* https://help.prusa3d.com/en/article/variable-layer-height-function_1750
* https://blog.prusaprinters.org/smooth-variable-layer-height-awesome-supports-slic3r-prusa-edition_5452/
* https://www.youtube.com/watch?v=oFIGMV-3GWU

alidorauxier, 2021-01-24:
I like this idea. Compared to variable layer height, think the difference would be that the 'substep' layers would only be used on the external perimeter and only on segments where the layer  is offset enough from the last one to benefit from the intermediate steps.
This would avoid printing the entire layer multiple times and concentrate only on the shallow details.

Battlevaria, 2021-01-24:
alidorauxier is right. As you can see in my original post, I already use variable layer height to help on this, but it's a very coarse instrument. Having this only on the final layer would be a lot easier and faster.

starthal, 2024-03-22:
> I expect printing the thinner layers first would work best, avoiding bringing a hot nozzle down close to a higher layer.

On the other hand, this almost describes how "ironing" works today, correct? Intentionally return the hot nozzle to an already-printed layer and extrude a bit more while passing over it.

As I understand it, ironing does happen on these "stair steps", but it uses the same method as on the topmost surface.

[image]

In this case, instead of returning to the same layer n, you could descend lower (between layer n-1 and layer n) and extrude an additional "antialias" perimeter. The new material along with the heat might give the desired smoothing.

(This would likely need to pair with an "exclusive" tolerance mode #8138 to avoid exceeding the model envelope)

AndrewFarley, 2026-09-05:
This is an important feature that is new and popular with various youtube videos around it, please keep this alive and implement this.  It's already been done in other slicers which are even forks of our slicer, so mabye it's easier than you think?  Especially with AI now?  :)

ColinTimBarndt, 2026-09-05:
@AndrewFarley Prusa wants to use Issues only for reproducible bug reports in the future and move feature requests into discussions. I've created one here: https://github.com/prusa3d/PrusaSlicer/discussions/15667

closed 2026-10-06 by Jan-Soustruznik