prusa3d/PrusaSlicer #5861: FR: Limited precision in variable layer height

Battlevaria, 2021-01-23:
### Version
2.3.0

### Operating system type + version
macOS Catalina Version 10.15.7

### 3D printer brand / version + firmware version (if known)
Prusa i3 mk3s, unmodified

### Behavior
_Describe the problem_

When using the variable layer height feature, the possible layer heights have 6 significant digits. My printer, though great, can't effectively print much finer than 0.01mm, so the last four digits are just noise. In fact, I would prefer to keep the layer heights to [magic numbers](https://3dprinterly.com/3d-printer-magic-numbers-getting-the-best-quality-prints/) for highest quality prints. 

_Current view_

[image]

_Desired functionality_

An advanced option of minimum step of variable layer height, presumably in the Printer settings, would make it easier to precisely select the desired layer height.

_Suggested view (mockup)_

[image]

_Is this a new feature request?_
Yes

rtyr, 2021-01-23:
Seems like a duplicate of https://github.com/prusa3d/PrusaSlicer/issues/202.

quetzalcoatl, 2021-01-30:
True, a duplicate, but much better explained IMO

edwardhorsford, 2022-06-29:
I'd also really like this - my use case is wanting the first two layers (and those layers only) to be 0.1mm, and thereafter all layers 0.2mm. With the variable layer height feature, reducing the first to 'fades' out for other layers - meaning I can't be precise with the layer heights.

stepikovo, 2022-07-02:
@edwardhorsford for a precise modification like this, you can use the _Height range Modifier_ which can be found under right-click.

I'm closing this as a mentioned duplicate

closed 2022-07-02 by stepikovo

edwardhorsford, 2022-07-04:
@stepikovo thank you, I was not aware of this feature. This is exactly the sort of thing I was looking for.

Battlevaria, 2022-07-06:
It seems one can only specify one Height range Modifier, that limits the feature quite a bit.

stepikovo, 2022-07-10:
@Battlevaria In the Object list, select the Layers modifier and then click on the + in the Height ranges.