prusa3d/PrusaSlicer #5862: Feature Request: Add "change STL" function that works like "reload from disk" but from a different STL file

adamdport, 2021-01-23:
### Version
2.3.0+

### Operating system type + version
Mac OS 11.1

### Behavior
It's currently impossible (afaik) to update a Prusaslicer project with a new version of the same model. Say version 1 is a cube. I create a Prusaslicer project, add modifiers, and paint on some seam enforcers. I then modify the cube (eg. same size, just has a hole) creating version 2. How can I apply the existing modifiers and seam enforcers to version 2?

One workaround is described in #2517–just copy/paste the modifiers into the new model. Unfortunately, modifier positions aren't applied if the model is different (though it does seem to work if the model is identical). Secondly, seam enforcers/blockers can't be copied/pasted in the same way.

Another workaround is to overwrite the STL file that was used to create the Prusaslicer project with the second version, and then "reload from disk". This works OK, but is very cumbersome, especially since it requires that version1.stl be destroyed.

My suggestion is to add a "change model" function that works like "reload from disk" but from a different stl and/or path.

_Is this a new feature request?_
Yes

quetzalcoatl, 2021-02-22:
@adamdport I work around it by renaming files. I create an initial project/settings and load an STL file called, for example, "model.stl" and then immediatelly I rename it to "model.v001.stl". Project remembers "model.stl" and it does not exist anymore. Then I move on and export various version of the same model under other files names, like, "model.v002.stl", "model_large.foo.bar.10.stl" and so on. When I want to reslice with reload/replace, I just rename "model_large.foo.bar.10.stl" to "model.stl", press reload, and 'undo' the rename afterwards. Just a tiny little nuisance, but works for me so far.

adamdport, 2021-02-22:
Ha, nice hack. I have to imagine this is a fairly common workflow for anyone who designs parts and needs to do anything at all extra in the slicer, seems like this should be far more polished, but I might give that a try meanwhile. Thanks for sharing!

MNeMoNiCuZ, 2021-09-07:
I would love this feature to be properly implemented as well.

If the "Reload From Disk" fails, it should let you know and ask if you want to browse for an alternative file to replace it with.

This is common behavior in Maya and Unreal Engine for example.

lukasmatena, 2021-09-07:
@MNeMoNiCuZ 
> If the "Reload From Disk" fails, it should let you know and ask if you want to browse for an alternative file to replace it with

This seems to be a different request than @adamdport 's. Anyway, both should be solved in [PrusaSlicer 2.4.0-alpha1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1). You can give it try and report if all is fine.

Closing.

closed 2021-09-07 by lukasmatena