prusa3d/PrusaSlicer #5867: Slicer not detecting possible collision after MMU fails to load

fpvdoc, 2021-01-24:
### Version
Version 2.3.0+win64

### Operating system type + version
Win10_64

### 3D printer brand / version + firmware version (if known)
Prusa MK3s+MMU2s
### Behavior
After a MMU load fail the extrudr will move to the left side of the bed. Purge block is on the upper right corner. On it's way to the fail position the extruder slightly touched the printed part. on a higher position in a print, that would've be a crash into the model.

* _Steps needed to reproduce the problem_
Force a loadfail in the MMU at layer height of >= 41
* _Expected Results_
Extruder moves to the left corner and crashes into the printed part even though prusa slicer didn't detect a possible collision
* _Actual Results_
 See picture attached
_Is this a new feature request?_
Not sure if there is an issue already. If so, please just merge.

Why is the extruder not just going straigt up in Z above the purge block. There is no benefit moving to the left and even worse if the slicer doesn't proof for possible crashes after mmu load fail.

[Design3.zip](https://github.com/prusa3d/PrusaSlicer/files/5862231/Design3.zip)
[image]

rtyr, 2021-01-24:
This is completely handled by the firmware.

I suppose it can only happen with experimental "no sparse layers" option enabled.

> Why is the extruder not just going straigt up in Z above the purge block. There is no benefit moving to the left and even worse if the slicer doesn't proof for possible crashes after mmu load fail.

As I mentioned above, this is handled by the firmware. You can open issue here https://github.com/prusa3d/Prusa-Firmware/issues.

rtyr, 2021-05-05:
I am closing this issue, it is firmware related (also duplicate of https://github.com/prusa3d/Prusa-Firmware/issues/2540).

closed 2021-05-05 by rtyr