prusa3d/PrusaSlicer #5868: Can't connect to SL1 from PrusaSlicr 2.3.0 (final release)

RobMeades, 2021-01-24:
### Version
2.3.0+win64 final release version

### Operating system type + version
Windows 10 (20H2)

### 3D printer brand / version + firmware version (if known)
SL1 (SW version 1.4.2)

### Behaviour
Having updated to 2.3.0 (release version) but I am no longer able to connect to my SL1: the `test` button and all attempts to use the `G` button hit the 10 second timeout.  I know the IP address is correct, the SL1 is switched on and displays the same IP address (as does my router), I have port 8000 (I've also tried without the port number added for good measure) and the API key is that which the SL1 displays under its Advance Settings->Login Credentials menu.  The username is unchanged from the default (`maker`).  And obviously I have connectivity from the computer to the router as I am using it to post this.  And connectivity worked fine with the previous version of PrusaSlicr which I last used a day or two ago.

What might I be doing wrong?

[image]

#### Project File (.3MF) where problem occurs
[tesla_cube_inlet_nozzle_casing.zip](https://github.com/prusa3d/PrusaSlicer/files/5862393/tesla_cube_inlet_nozzle_casing.zip)

tamasmeszaros, 2021-01-26:
Could you please run PrusaSlicer from the command line with the parameter `--loglevel 9` and post the output here? Thanks in advance.

RobMeades, 2021-01-26:
Hi and thanks for responding.  I've done that but where do I find the log output afterwards?

```
C:\Program Files\Prusa3D\PrusaSlicer>prusa-slicer.exe --loglevel 9

C:\Program Files\Prusa3D\PrusaSlicer>dir
 Volume in drive C is Windows
 Volume Serial Number is 343E-1A59

 Directory of C:\Program Files\Prusa3D\PrusaSlicer

24/01/2021  13:44    <DIR>          .
24/01/2021  13:44    <DIR>          ..
11/01/2021  14:23           543,027 libgmp-10.dll
11/01/2021  14:23           436,011 libmpfr-4.dll
24/01/2021  13:44    <DIR>          mesa
11/01/2021  14:23           590,112 msvcp140.dll
11/01/2021  14:23           197,632 prusa-gcodeviewer.exe
11/01/2021  14:23           187,392 prusa-slicer-console.exe
11/01/2021  14:23           187,392 prusa-slicer.exe
11/01/2021  14:23        39,171,072 PrusaSlicer.dll
24/01/2021  13:45    <DIR>          resources
24/01/2021  13:45           141,781 unins000.dat
24/01/2021  13:44         1,199,947 unins000.exe
11/01/2021  14:23           101,872 vcruntime140.dll
11/01/2021  14:23            44,528 vcruntime140_1.dll
              11 File(s)     42,800,766 bytes
               4 Dir(s)  545,662,349,312 bytes free

C:\Program Files\Prusa3D\PrusaSlicer>cd C:\Users\...\AppData\Roaming\PrusaSlicer

C:\Users\...\AppData\Roaming\PrusaSlicer>dir
 Volume in drive C is Windows
 Volume Serial Number is 343E-1A59

 Directory of C:\Users\...\AppData\Roaming\PrusaSlicer

26/01/2021  09:59    <DIR>          .
26/01/2021  09:59    <DIR>          ..
26/01/2021  09:59    <DIR>          cache
30/12/2020  22:48    <DIR>          filament
24/01/2021  13:04    <DIR>          physical_printer
30/12/2020  22:49    <DIR>          print
06/06/2020  20:18    <DIR>          printer
26/01/2021  09:59             3,300 PrusaSlicer.ini
31/12/2020  14:42    <DIR>          sla_material
25/01/2021  00:55    <DIR>          sla_print
24/01/2021  13:17    <DIR>          snapshots
01/06/2020  22:19    <DIR>          vendor
               1 File(s)          3,300 bytes
              11 Dir(s)  545,641,795,584 bytes free
```

RobMeades, 2021-02-14:
Mysteriously, I go to try it again and it is now working :-).

closed 2021-02-14 by RobMeades