prusa3d/PrusaSlicer #5869: Printer profile is missing after crash

Zimer, 2021-01-24:
### Version
_Version of PrusaSlicer used goes here_
Version 2.3.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
MK3S+

### Behavior
After PrusaSlicer crashed, I can't see my saved printer profile that is configured to work with Octoprint. (The profile name is: Octoprint + Prusa I3....)
[image]

However, it seems still that the profile is there because when I'm trying to create a new one, with the same name as the older one, I'm getting this error:
[image]

rtyr, 2021-01-26:
Could you please zip and send us your complete configuration directory so we can have a look? (Help -> Show Configuration Folder).

rtyr, 2021-03-10:
No response. Closing.

closed 2021-03-10 by rtyr