prusa3d/PrusaSlicer #5870: Non-responsive moving between monitors of different resolutions

sworcester001, 2021-01-24:
### Version
_Version of PrusaSlicer used goes here_2.3.0

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_ Windows 10 Pro Build 1904
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_ N/A

### Behavior
* _Describe the problem_ Non-responsive moving between monitors of different resolutions
* _Steps needed to reproduce the problem_ dual mionitors, one 4K, one 1080, drag the instance between the two screens
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_ moves with cursor
* _Actual Results_ stops about halfway between and has to catchup several seconds later
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_no
#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

foreachthing, 2021-01-24:
My monitor setup is:
1. 1680 x 1050
2. 1920 x 1200

OS: Win10 (Version 20H2, Build 19042.746) 
GPU: GeForce RTX 2070
Driver: 460.89 (Studio)

No issue here.

sworcester001, 2021-01-24:
I figured it out, it doesn't like text scaling. If I bring the scaling down to 100%, it acts as expected.
I will have to learn to squint until it is fixed

closed 2026-09-29 by prusaslicer-bot[bot]