prusa3d/PrusaSlicer #5871: bridge_angle=0 chooses longest path for flat overhangs

jgreely, 2021-01-24:
### Version

2.3.0+

### Operating system type + version

macOS Catalina 10.15.7

### 3D printer brand / version + firmware version (if known)

Prusa i3 MK3S (error visible in Preview)

### Behavior
* For objects with horizontal overhangs, bridge_angle=0 chooses long paths that result in significant sag.
* _Steps to repeat:_
  * Add a small model with flat overhangs, such as [this one](https://dotclue.s3.amazonaws.com/prusa/bridge_angle_auto_test.stl).
  * Select MK3S printer, Prusament PLA filament, 0.20mm QUALITY; enable supports on build plate.
  * Slice.
  * Preview bridge infill layer (should be layer 58).
* _Expected Results_
  * Infill should use short path (45 or 90 degree).
* _Actual Results_
  * Infill uses longest path. If you scale the model to 150% or larger, the far overhang is at a slight angle, while the near one remains worst-case.
[image]

_Is this a new feature request?_

No.

#### Project File (.3MF) where problem occurs

[bridge_angle_auto_test.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/5862801/bridge_angle_auto_test.3mf.zip)

EmilyBjoerk, 2021-03-21:
This has been bothering me as well. In addition to the above bug report, trying to rotate the model to control the bridging direction also doesn't work. I.e. if you rotate the model 90 degrees around z-axis the bridging direction stays the same.

EmilyBjoerk, 2021-03-21:
Seems like a dupe of #4797

closed 2026-09-29 by prusaslicer-bot[bot]