prusa3d/PrusaSlicer #5872: PrusaSlicer not remembering last location used on saving, exporting, loading...

batata004, 2021-01-24:
Hi, I am using the Windows version of latest stable build of PrusaSlicer. Everytime I load an STL, a configuration, or everytime I save to my SD card the GCODE generated, the software never remembers the last directory used in each action. It always Defaults to my directory. Using Simplify3D and Cura I never had this problem.

I already went to PREFERENCES and made sure the remember output directory was checked and this problem still happens. It's very annoying because every single time I save a GCODE to my SD card and later I plug it back to save another GCODE, the software always defaults to my Desktop folder... save for loading STL and loading config file.

neophyl, 2021-01-24:
It remembers the last location on the 4 windows machines I have access to.  One is still a windows 7 machine and the other 3 are on windows 10. 
Are you running with some form of restricted windows account possibly ?  If you run as an admin level account does it then work ?

batata004, 2021-01-24:
@neophyl My friend, I am running latest Windows 10 (64 bits) logged as administrator (I dont have other user accounts, just admin) and I installed the software directly from the Prusa website clicking on the 1st option of windows.

This problem does not happen with Cura nor with S3D. I recorded a video demonstrating this problem happening 2 times so you can see it happening:

https://photos.app.goo.gl/qs8p8bcc8efgGAGA8

If you have any questions, I am here to provide information as much as I can.

neophyl, 2021-01-25:
Well on windows Prusa stores that information in its PrusaSlicer.ini file.  For most users with a default install that will be in C:\Users\<_username_>\AppData\Roaming\PrusaSlicer 
If the software cant write to the file then it would have issues.  Which is why I asked if you had some form of restricted account.

I'm sure a dev will be along at some point to help you out but until then you might want to take a look at that ini file and see whats in there.

batata004, 2021-01-25:
Thank you for helping me into this. I searched for the file you told me and couldnt find it. For some reason that file is not being generated on my Windows when running PruseSlicer. :( That's a pitty because this software is amazing, the level of control on the GCODE is huge compared even to S3D.

While this problem is not fixed I will keep using S3D cause it consumes lots of times everytime having to browse the exact place to save, load.. files.

foreachthing, 2021-01-25:
> @neophyl My friend, I am running latest Windows 10 (64 bits) logged as administrator (I dont have other user accounts, just admin)... 

MHO: create a new non-admin user. Try it with that new account. No need to be admin all the time. Again: my opinion!

batata004, 2021-01-25:
I think the problem is because of some "permission bug" that PrusaSlicer does on Windows installation. I am logged as admin and everytime I try to copy a file to the prusaslicer directory I am requested to login as admin but this does not happen with any other software I have installed on my program files folder on windows.

PrusaSlicer is not a banking app, it does not handle confidential information... I see no point it being installed on a folder with restrictive permission. And the big problem here is that even running the program on Windows with admin privilege, it still does not create the file "PrusaSlicer.ini".

@foreachthing I dont think your suggestion would help since I am logged with Admin and any other user account I create will have less privilege.

neophyl, 2021-01-25:
I think @foreeachthing's point is that it sounds like PS is having some issue with your configured account.  There are huge numbers of people using it on Windows who have not experienced the issue you are having.  Its also not a case of needing extra permissions as its installed in a perfectly normal place already and the roaming folder is where most software puts their settings.

If you create a new account and try installing it under that account it would be good to know if it works or not.  

It may be that there's something that could be done differently on the installer side but an isolated case with no information as to why that account is not allowing permissions does not give the developers anything to go on.  

If you truly are logged in as an admin then copying a file to that folder using explorer should not pop up ANY permission issues.  Have you checked what permissions are set on the Prusa folder in Roaming compared to one of the other folder located there ?  If you try and copy a file to a different folder in roaming do you also get a permissions window ?

batata004, 2021-01-29:
Guys, I tried everything and nothing solved the problem :( The most strange thing is that S3D works perfectly and Slic3r too. Anyway, since nobody else is reporting this problem, I will close this, maybe is something related to my windows environment. Thanks for trying help me

closed 2021-01-29 by batata004