prusa3d/PrusaSlicer #5873: PrusaSlicer is not slicing properly....

enigmatic-system, 2021-01-24:
### Version
2.3.0

### Operating system type + version
OS: Arch Linux x86_64
Kernel: 5.4.91-1-lts

### 3D printer brand / version + firmware version (if known)
Ender 3 but this is occurring without a printer.

### Behavior
Prusa-Slicer is slicing automatically and does not give an option to modify settings before slicing like in 2.2.0. 
As soon as I load an STL from a download, Blender, or Cinema4d the objects get sliced and changed settings are not taken into account.
It just seems like a bug and it confuses me and is fucking up my prints >:^)

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSli
[projectFile.zip](https://github.com/prusa3d/PrusaSlicer/files/5863330/projectFile.zip)
cer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_
[image]
picture of a simple cube with the brim turned on
[image]
picture of the settings for the simple cube

I can successfully do all of this on 2.2.0 on my other laptop that has an older version.

I've uninstalled, reinstalled; downgraded, tried to mess with libs a bit but this just seems like a bug.
I'll continue to mess around with it but here's some info for now~

rtyr, 2021-01-28:
It it your custom build?

> Prusa-Slicer is slicing automatically.

I guess you have the "Background processing" enabled in the preferences?

Anyway, I am unable to reproduce it.

rtyr, 2021-04-22:
No response. Closing.

closed 2021-04-22 by rtyr