prusa3d/PrusaSlicer #5876: se apaga el portatil al usar el programa

javitec, 2021-01-25:
### Version
 
2.3.0

### Operating system type + version

windows 10

### 3D printer brand / version + firmware version (if known)

para un perfil de impresora Anet A8 modificada

### Behavior

* _Describe the problem_

el problema que estoy presentando es el siguiente. descargue la versión 2.3.0 de pruslicer para windows 10.
primero con todas las opciones hasta los drivers, inicia normalmente modifico un perfil para impresora anet a8 pero al momento de escoger un stl para hacer un laminado en cualquier momento apaga mi pc por completo, esto sucede aleatoriamente es decir puedo hacer un laminado de algún archivo y se apaga o alcanzo hacer el laminado y al querer hacer otro laminado igual se apaga por completo el pc, no tengo ningún otro programa en función solo el laminador por ende es el que me hace apagar el portátil... quisiera saber a que se debe y como podría solucionarse. muchas gracias.

javitec, 2021-01-25:
PrusaSlicer
Version:   2.3.0+win64
Build:     PrusaSlicer-2.3.0+win64-202101111315

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 19042), 64-bit edition
Total RAM size [MB]: 8,470MB
OpenGL installation
GL version:   4.6.0 NVIDIA 391.35
Vendor:       NVIDIA Corporation
Renderer:     GeForce GT 635M/PCIe/SSE2
GLSL version: 4.60 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_gpu_shader_int64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_indirect_parameters
[... 245 lines cut ...]
GL_NV_texture_rectangle
GL_NV_texture_rectangle_compressed
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

burner-, 2021-01-29:
slicing is very cpu intensive process. If your laptop fan, psu or battery does not work properly it can crash it. Anyway it is not fault of software, it is fault of mailfunctioning hardware. You can try help/workaround this issue by setting affinity for process via task manager so that it can use only one core. But right solution for it is fix your hardware.

rtyr, 2021-02-18:
This is almost certainly hardware issue. Closing.

closed 2021-02-18 by rtyr