prusa3d/PrusaSlicer #5878: First layer of support.

AKlionRU, 2021-01-25:
How can you make sure that the first layer of support does not crawl so far beyond the contours of the model.
[image]

rtyr, 2021-01-25:
There are already similar issues opened, like https://github.com/prusa3d/PrusaSlicer/issues/3772. We will likely extend the settings. Closing as a duplicate.

closed 2021-01-25 by rtyr

batata004, 2021-02-14:
So how do I remove the zig-zag/raft from the support? It's too annoying and is completelly useless on 95% of cases, just wastes plastic, energy and printing time. This issue is almost 1 year old and no news about it? If it will not be implemented - FINE - but please let us know that this will not be fixed.