prusa3d/PrusaSlicer #5879: REQ: Draw line in seam paint tool

AndKe, 2021-01-25:
The seam paint tool is great.
In order to paint a straight seam across many levels, it's hard to draw a perfectly straight line.
Please add an option (when shift is pressed) to draw a straight line from the last point.
Also, an option to snap to some angles.  (45 .. 60 .. 90°)

Thank you.

### Version
2.3.0

### Operating system type + version
Ubuntu 

### 3D printer brand / version + firmware version (if known)
n/a

### Behavior
n/a
#### Project File (.3MF) where problem occurs
n/a

bttguy, 2021-02-05:
+1, looking forward to it!

rtyr, 2021-02-05:
Same request as https://github.com/prusa3d/PrusaSlicer/issues/5970. I will close this one to reduce duplicates.

closed 2021-02-05 by rtyr