prusa3d/PrusaSlicer #5880: PrusaSlicer-2.3.0+linux-x64-202101111322 crashes upon slicing a big STL file

ekatus, 2021-01-25:
### Version
PrusaSlicer-2.3.0+linux-x64-202101111322

_
### Operating system type + version
Linux 5.4.0-58-generic #64-Ubuntu SMP Wed Dec 9 08:16:25 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux
on an Intel NUC6CAYH with 8GB RAM

### 3D printer brand / version + firmware version (if known)
Custom profile for FLSUN Q5 Delta printer

### Behavior
PrusaSlicer crashes upon slicing a big lithopane (not included for data privacy issues, showing a person's face) with a file size of about 386,1MB. The file itselfs loads and preview of the model is rendered.

PrusaSlicer started as an AppImage from both the desktop and the console. Console shows error messages that seem unrelated to the problem. Upon crash, a small popup is openend but not fully rendered and in a split second, the process is killed ("getötet" output in console) No futher information about the reason of the crash.

--
(prusa-slicer:834710): GLib-GObject-CRITICAL **: 12:03:06.924: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:834710): GLib-GObject-WARNING **: 12:03:06.958: invalid (NULL) pointer instance

(prusa-slicer:834710): GLib-GObject-CRITICAL **: 12:03:06.958: g_signal_handlers_block_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed

(prusa-slicer:834710): GLib-GObject-WARNING **: 12:03:06.958: invalid (NULL) pointer instance

(prusa-slicer:834710): GLib-GObject-CRITICAL **: 12:03:06.958: g_signal_handlers_unblock_matched: assertion 'G_TYPE_CHECK_INSTANCE (instance)' failed
Getötet

--

Legodev, 2021-02-20:
Have the same issue, with my own profiles as well as the default ones.

It dies with a segmentation fault.

bubnikv, 2021-02-22:
we cannot reproduce your issue without a 3MF file.

DomiStyle, 2021-02-27:
@bubnikv PrusaSlicer-2.3.0+linux-x64-GTK3-202101111352.AppImage crashes for me (Segmentation fault) on Linux when slicing either one of these STL:
https://www.thingiverse.com/thing:4220523

MasterCATZ, 2021-03-05:
this also  seg faults for me on Ubuntu 20.04 , with Vega 64 

using both app image and built from source 

[P047_Belt.zip](https://github.com/prusa3d/PrusaSlicer/files/6089576/P047_Belt.zip)

https://www.thingiverse.com/thing:4699673

Legodev, 2021-03-08:
> LIBGL_ALWAYS_SOFTWARE=true

I can confirm, this solves it for me as well. Its very slow, but does not crash.

jjoakim, 2021-03-19:
I also have a lot of crashes when slicing is used. One workaround I found was that if I sliced a working object first and then lowered the level so only a handful of slices at the bottom is showed. When I go back into the model view mode I can add more complex models and slice them. But when in slicing mode and if I raise the number of  levels visible the app will eventually crash. So from my experience it is not the slicing itself that is broken but rather the display of the sliced model.

As LIBGL_ALWAYS_SOFTWARE=true works as a workaround as well it seems like the problem could potentially originate from mesa.

mixicz, 2021-03-21:
I have exactly same problem. 
GPU Vega 20 (Radeon VII), drivers mesa 21.0.0-1, distro Debian testing with GPU drivers from Debian experimental.
`LIBGL_ALWAYS_SOFTWARE=true` works as workaround

Szum123321, 2021-03-28:
I'm having the same issue on Arch with kernel 5.11.8-arch1-1 and AMD Ryzen 5 3500U. Slicer works with test cube but crashes once I try to view more complex model.

ksergey, 2021-04-03:
Same for me. Arch with kernel 5.11.11-arch1-1 and Ryzen 5 PRO 4650U

kletterstein, 2021-04-14:
And same for me. Kubuntu 20.10 and Radeon RX 580. Software rendering also works as a work-around, but viewing the sliced model is extremely slow. The problem occurs with all models since a few days.

lukasmatena, 2021-04-15:
Can you guys please try out the new [2.3.1 patch release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1-rc) to see if it helps? It has in https://github.com/prusa3d/PrusaSlicer/issues/6345#issuecomment-820503201. Thanks.

jjoakim, 2021-04-15:
> Can you guys please try out the new [2.3.1 patch release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1-rc) to see if it helps? It has in [#6345 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/6345#issuecomment-820503201). Thanks.

I tried again with the current version and this time I could open my project and slice it without problems. My first guess was that a driver (mesa) update had fixed the problem. I then made a lot of copies of my object and sliced it which also worked. But then I started to spin the view around for a short while and the program crashed. I repeated the same procedure with the release candidate version. It also crashed in the same manner. Spinning a single object did not seam to make it crash. My object is not that big. Its dimensions are 26.28x14.450 mm.

kletterstein, 2021-04-15:
> Can you guys please try out the new [2.3.1 patch release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1-rc) to see if it helps? It has in [#6345 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/6345#issuecomment-820503201). Thanks.

I checked with my last two STL files and now I could slice without problems. Great, thanks a lot!

turtle-bazon, 2021-04-16:
> Can you guys please try out the new [2.3.1 patch release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1-rc) to see if it helps? It has in [#6345 (comment)](https://github.com/prusa3d/PrusaSlicer/issues/6345#issuecomment-820503201). Thanks.

Yeah. Version 2.3.1-rc works fine AppImage on ArchLinux. Previous 2.3.0 AppImage crashes.

christoph-k, 2021-04-16:
Tried 2.3.1-rc (Appimage) on ArchLinux. 
With 2.3.0 i had crashes no matter what i was slicing.
With 2.3.1-rc i can slice a couple of smaller/less complex objects but there are still a lot of crashes.
https://www.thingiverse.com/thing:903411 crashes any time i try to slice it.

turtle-bazon, 2021-04-22:
On my system 2.3.1 slice that without problems. I use AppImage for GTK3 on ArchLinux.

christoph-k, 2021-04-24:
2.3.0 on ArchLinux actually seems to work again. I guess this is due to some updated dependencies.

turtle-bazon, 2021-04-26:
I can confirm that with all updates 2.3.0 on ArchLinux works on models that previous crashed.

closed 2026-09-29 by prusaslicer-bot[bot]