prusa3d/PrusaSlicer #5884: Slicer 2.3.0 Win10 Virus in prusa-slicer.exe

rwensley, 2021-01-25:
### Version
_2.3.0+win64-2021-1111315
_Use `About->About PrusaSlicer` for release versions_    **Can not install**

### Operating system type + version
_Windows 10 Enterprise

### 3D printer brand / version + firmware version (if known)
_Can not install

### Behavior
Upon installation, the corporate IT installed virus protection, Symantec Endpoint Protection, deletes the prusa-slicer.exe file.
Symantec reports "There is strong evidence that this file is untrustworthy.
[image]
Symantec is version 14.2 RU2 MP1 build 5569 (14.2.5569.2100)

To reproduce simple try to install prusa-slicer with this version of Symantec.

rtyr, 2021-01-26:
Please report this incident to the Symantec, it needs to be fixed on their side.

Closing.

closed 2021-01-26 by rtyr

markhedinger, 2021-07-25:
Actually, you should be reporting this to Symantec. You have multiple requests about this issue and you just keep closing them, blaming symantec and your users.

At the end of the day, this is your software and you want us to use it. So you get in contact with symantec and sort it out, because we have plenty of other options next to prusa. 

At the end of the day most of us have more trust in Symantec than Prusa, so if symantec says your software contains threats and viruses, I'm gonna go with that and absolutely (and immediately) uninstall your files. 

Fix your software.