prusa3d/PrusaSlicer #5889: PrusaSlicer segfaults immediately after slicing complex objects

tullo-x86, 2021-01-26:
### Version
Version 2.3.0+linux-x64 (AppImage)

### Operating system type + version
Arch Linux, up to date as of 2021-01-25 00:08Z

#### System Info
```
PrusaSlicer
Version:   2.3.0+linux-x64
Build:     PrusaSlicer-2.3.0+linux-x64-202101111322

Operating System:    Unix
System Architecture: 64 bit
System Version:      Linux 5.10.10-arch1-1 x86_64
Total RAM size [MB]: 33,535MB
OpenGL installation
GL version:   4.6 (Compatibility Profile) Mesa 20.3.3
Vendor:       AMD
Renderer:     AMD SIENNA_CICHLID (DRM 3.40.0, 5.10.10-arch1-1, LLVM 11.0.1)
GLSL version: 4.60
Installed extensions:

GL_AMD_conservative_depth
GL_AMD_depth_clamp_separate
GL_AMD_draw_buffers_blend
GL_AMD_framebuffer_multisample_advanced
GL_AMD_multi_draw_indirect
GL_AMD_performance_monitor
GL_AMD_pinned_memory
GL_AMD_query_buffer_object
GL_AMD_seamless_cubemap_per_texture
GL_AMD_shader_stencil_export
GL_AMD_shader_trinary_minmax
GL_AMD_texture_texture4
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ANGLE_texture_compression_dxt3
GL_ANGLE_texture_compression_dxt5
GL_APPLE_packed_pixels
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
[... 265 lines cut ...]
### Behavior
* Complex objects appear to slice, but as soon as the toolpath is displayed, PrusaSlicer crashes
  * Simple objects, or low-detail slices of the same model, appear to succeed

#### Reproduction steps
* Grab Benchy from the official repo
  * Both the [3MF](https://github.com/CreativeTools/3DBenchy/blob/master/Single-part/3DBenchy.3mf) and [STL](https://github.com/CreativeTools/3DBenchy/blob/master/Single-part/3DBenchy.stl) version can reproduce this issue
* Load Benchy in the editor, use auto-arrange to centre it if necessary
* Select "0.15mm QUALITY" print settings and hit "slice now"
  * **Observe:** Slice completes successfully, the plain Slic3r yellow preview is displayed for a moment, then the "feature type" preview loads and replaces it
* Select "0.05mm ULTRADETAIL" print settings and hit "slice now"

##### Expected
Similar to 0.15mm, the preview first loads the plain yellow toolpath, then the "feature type" view replaces it

##### Actual
The preview displays the plain yellow toolpath, and then PrusaSlicer crashes, reporting a segmentation fault.

LOG FILE: [prusaslicer-2.3.0-benchy-ultradetail.txt](https://github.com/prusa3d/PrusaSlicer/files/5870541/prusaslicer-2.3.0-benchy-ultradetail.txt) (obtained by `/PrusaSlicer-2.3.0+linux-x64-202101111322.AppImage --loglevel=9 | tee ~/prusaslicer-2.3.0-benchy-ultradetail.txt`)

##### Notes
This does not appear to occur on simpler objects, even if they result in a vary large G-code file. This was confirmed by adding a 200 x 200 x 200 cube to the empty plate and slicing it at "0.15mm QUALITY". I tried slicing the same cube at "0.05mm ULTRADETAIL", but ran out of memory before it finished slicing.

#### Project File (.3MF) where problem occurs
[Literally Benchy](https://github.com/CreativeTools/3DBenchy/blob/master/Single-part/3DBenchy.3mf)

lukasmatena, 2021-01-26:
The description is similar to #5745 and #5859....

tullo-x86, 2021-01-26:
Those do look similar. Curiously, both of them are also on AMD cards (I'm using `amdgpu` on an RX 6800). I wonder if it's a driver interaction with the preview's GL renderer?

smirgol, 2021-01-27:
I have a very similar problem #5745, but unfortunately your two solutions don't fix it for me. With software rendering it will outright crash upon start, with the DRI disabled it will still crash **after** slicing. And yes, it has to be something related to the Mesa/amdgpu (like 20 something) driver, as with an older version of it I didn't have that issue in the past. And with PrusaSlicer 2.2 it just works, but it's missing features, obviously.

MasterCATZ, 2021-03-05:
I have been having the same issue since I started using a vega 64 

appimage and built from source 

how can I launch with software rendering only ?

tullo-x86, 2021-03-05:
> how can I launch with software rendering only ?

Read the issue description, under "known workarounds"

cubiq, 2021-04-01:
Same issue, I have an AMD gpu too. Only fix for complex objects for me is to use LIBGL_ALWAYS_SOFTWARE=true (thanks @tullo-x86 ). Fortunately I have a good CPU but it's still painful.

tullo-x86, 2021-04-01:
@cubiq did you try `LIBGL_DRI3_DISABLE=true` as well? If it runs in that mode on your environment, you'll be able to use your GPU to render.

cubiq, 2021-04-02:
> did you try `LIBGL_DRI3_DISABLE=true` as well?

yeah I tried, doesn't work unfortunately

DavidBittner, 2021-04-09:
I'm having this issue as well on my swaywm setup. None of the listed fixes have worked, but I am able to slice very simple models as was also mentioned. I maxed out logging and it seems to crash immediately after refreshing extrusion paths.

smirgol, 2021-04-09:
@DavidBittner which graphic stack are you on? Mesa probably? These problems started for me after a certain version of Mesa and affects all versions of Prusa Slicer >=2.3 (including RCs and alpha/beta). 2.2 works well though, but it is outdated.

For me it always crashes when calling `glMultiDrawElements` in `GCodeViewer.cpp` method `render_toolpaths()`, there either must be something odd with the buffers or it might be a racing condition thing instead. Don't know for sure without compiling a debug version of Mesa from scratch, which isn't exactly a fun thing to do. 

I can slice some things, but most things just end up in a crash and there is no pattern besides that if it crashes once for a specific object and specific settings, it will always crash. But with other settings, e.g. other infill value or other layer height, it _might_ not crash.

DavidBittner, 2021-04-09:
You were right about it being mesa. I went ahead and downgraded mesa a bunch to try and narrow it down and the crashes actually did stop. It was at version mesa-21.0.0.-1 that they stopped happening. I guess I will just keep mesa a bit downgraded for now until an update comes out.

But either way, downgrading mesa seems to have fixed the problem for me.

lukasmatena, 2021-04-15:
Can you guys please try out the new [2.3.1 patch release](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.3.1-rc) to see if it helps? It has in https://github.com/prusa3d/PrusaSlicer/issues/6345#issuecomment-820503201. Thanks.

cubiq, 2021-04-15:
works for me now, thanks for the heads up @lukasmatena

tullo-x86, 2021-04-17:
@lukasmatena Unfortunately, the v2.3.1 RC did not completely solve the issue on my machine. Benchy can now be sliced, but other complex models still fail.

This plays into my suspicion that the underlying issue is a race condition.

smirgol, 2021-04-17:
I need to test it more before I can say for sure that it has fixed the crashes completely, but as of now it slices a couple of things that 100% crashed before. No crashes yet - that is really cool!

Out of curiosity, which commit did fix the issue?

cubiq, 2021-04-20:
damn, still got some crashes. much less than before though

vase mode is also a no-no as @smirgol noted. Also tried the latest version