prusa3d/PrusaSlicer #5890: Option to set the placement distances for Add instace function

spdrp, 2021-01-26:
### Version
PrusaSlicer 2.3.0+win64

### Operating system type + version
Windows 10 x64 v20H2 build: 19042.746

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S MMU2S and SL1

### Behavior
Add instance [+] function adds new instance correctly, but to a fixed distance. This distance is X=5.9mm,Y=5.9mm (strange distance) for SL1, and X=12.5mm,Y=12.5mm (1/8 of 100mm, but still not a nice number) for MK3S.

### New feature request
Option to set these distances.

fe60, 2021-01-26:
Are you aware of the adjustable spacing when using Auto-arrange?
https://help.prusa3d.com/en/article/auto-arrange-tool_1770

spdrp, 2021-01-26:
> Are you aware of the adjustable spacing when using Auto-arrange?
> https://help.prusa3d.com/en/article/auto-arrange-tool_1770

Yes, I know about this function. It affects the auto-arrange and fill bed with instances, but doesn't affect the Add instance. The auto-arrange is not the best option for some layouts.

closed 2026-09-29 by prusaslicer-bot[bot]