prusa3d/PrusaSlicer #5891: Incorrectly displays g-code sliced in other slicers

AndylgTom, 2021-01-26:
### Version
PrusaSlicer G-code Viewer-2.3.0

### Behavior
PrusaSlicer G-code Viewer incorrectly displays g-code sliced in other slicers - Cura, Simplify3D. At the same time, they are displayed normally in Repetier-Host.
[image]

neophyl, 2021-01-26:
How about attaching the problematic gcode here so the devs actually have something to work with as an example ?

AndylgTom, 2021-01-26:
Example g-code
[example.zip](https://github.com/prusa3d/PrusaSlicer/files/5871878/example.zip)

rtyr, 2021-01-26:
What do you mean exactly, the bed size? This is intentional if the bed size is unknown. I was able to find the bed size in the S3D comments, so it can be likely improved, but I was not able to find that info in the g-code from Cura.

rtyr, 2021-01-26:
Related S3D comments:
```
;   strokeXoverride,180
;   strokeYoverride,180
;   strokeZoverride,180
```

closed 2026-09-29 by prusaslicer-bot[bot]