prusa3d/PrusaSlicer #5892: Printer stops mid print usind Slicer 2.3 for Prusa mini

marvinAmueller, 2021-01-26:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

rtyr, 2021-01-26:
I don't think this is related to the PrusaSlicer. It is likely some reading error (related to flash drive, filesystem or Octoprint) or firmware issue. 
Anyway, please provide the gcode and the information, where the problem happens (height).

marvinAmueller, 2021-01-26:
I tried it with a different flash to check the possibility. I sent it directly from Fusion360 to the slicer and exported it to stl beforehand.  I did not use octoprint yet. tried it with different prusament, tried it with and without filament sensor
- preparing the print with slicer 2.3 the all tries failed
- preparing the print with previous slicer version all different tries were ok
I will check if I still have the file and provide it

rtyr, 2021-01-26:
I am afraid we can't do anything without the file. I still don't think that this problem is PrusaSlicer related, be we can check the file or try to print it.

rtyr, 2021-03-10:
No response, Likely not PrusaSlicer related. Closing. 

Feel free to provide the g-code if you find it.

closed 2021-03-10 by rtyr

marvinAmueller, 2021-03-12:
thanks for following up.

haven't had the issue with the updated old slicer version and the new slicer for many prints now.

BR and a good weekend.

Marvin