prusa3d/PrusaSlicer #5893: Fan ramp not correct for layer 1

ccomtest, 2021-01-26:
### Version
Prusaslicer 2.3.0

### Operating system type + version
Windows 10, 2004 (B19041.746)

### 3D printer brand / version + firmware version (if known)
Prusa MK3S - Black PSU, FW 3.9.2

### Behavior
If you set up full fan at layer x (here layer 4) with a minimum speed (here 40%)  and you slice a model, 
you will have full mimum fan speed at layer 1 and layer 2 has the ramped goind up on layer 3 and 4.

So in the slicer it looks like, you have full minimum speed, then goes down as expected (fan ramp up) and then goes up again.
I think it should start with the expected fan speed at layer 1 when ramped (40/4 here = 10%) and not full fan speed and then go down to 20% at Layer 2.

This issue is with all models and all fan settings if you try to ramp it up with layers.

#### Project File (.3MF) where problem occurs
[image]
[image]
[image]

rtyr, 2021-01-26:
The values doesn't match with the screenshot, but I was able to reproduce it.

ccomtest, 2021-01-26:
Yes, sorry.
I have chosen some easier values for calculation. They should be examples only. The screenshots are only to show the issue.

Good it was to reproduce :)

bubnikv, 2021-02-08:
Fixed with https://github.com/prusa3d/PrusaSlicer/commit/de32992b4937a8ff6c9a6fc5f38ca1d092ec4787
Closing

closed 2021-02-08 by bubnikv

ccomtest, 2022-02-27:
Hi there,

now a similar issue seem to be in 2.4.0 (final).
When setting Fan to 100 at layer two and fan off for first layers to 0 it still has fan at 0 for first layer. For my understanding when I set 100% at layer two without fan off for first layers PS should slice with 50% at layer 1. Which is not the case, the fan is off in preview and also at print

Best regards
Peter