prusa3d/PrusaSlicer #5895: Slicing may fill holes in objects

starkjohann, 2021-01-26:
### Version
PrusaSlicer-2.3.0+

### Operating system type + version
macOS 11.2

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK2

### Behavior
When I slice the following object:

[image]

the hole in the object is filled from the edge upward (after inserting a bridging layer):

[image]

The attached zip file contains:
* An OpenSCAD file for the object.
* An STL file created from the object.
* The screen shots from above
* A Slicer project (.3mf file) with all settings

The bug vanishes if some parameters are changed slightly. See comments in the OpenSCAD file.

[PrusaSlicerBugTest.zip](https://github.com/prusa3d/PrusaSlicer/files/5872615/PrusaSlicerBugTest.zip)

rtyr, 2021-01-26:
The issue will likely dissappear if you repair the model. OpenSCAD does have quite a history of generating problematic models like this (see the issues with "model import" label). We should improve the PrusaSlicer to handle such problematic STL better though.

starkjohann, 2021-01-26:
Version 1.41 imports this model without any issues. I did not want to test with 2.2.* because it would have reset my settings.

Would be great if PrusaSlicer would apply such repairs automatically, because it's very hard to find all those issues in the sliced preview.

rozi-risq, 2021-01-28:
> Would be great if PrusaSlicer would apply such repairs automatically, because it's very hard to find all those issues in the sliced preview.

Yes, I second this. The sliced model kind of hit and miss in Prusa Slicer. Sometimes it perfectly sliced, but other time not. Imagine if you have a very complex object, you need to look each area carefully.

starkjohann, 2021-01-28:
I never had this problem in previous versions of Slicer. It sometimes reported that the model had to be repaired. Has this code been removed? Or are there new optimizations which require a stricter model conformance? If it's the latter, is there an option to disable the new code?

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo

bubnikv, 2021-07-02:
Fixed with PrusaSlicer 2.3.2-beta.