prusa3d/PrusaSlicer #5896: My project settings are not being saved/loaded correctly

pigskins65, 2021-01-26:
### Version
2.3.0+win64

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa M3KS+ stock no mods
3.9.3 

### Behavior
I recently upgraded from 2.2 to 2.3. I have a large number of project files saved from 2.2 and I use a pretty consistent group of settings across the Print, Filament, and Printer tabs. They are usually all the same changes except for a small number where I have slowed down the first layer speed.

After upgrading I found that when I open any project file from 2.2 in 2.3 there are 6 screens where the settings are different from what I saved in 2.2. I have 6 screenshots that I will attach. What I have been doing is going through each screen after loading a 2.2 project file and clicking the orange back arrow to reset whatever it is back to its previous setting except for the settings I know I changed myself such as infill pattern, top/bottom layers, filament extrusion multiplier, filament color, and first layer speed if I had changed it. I then save this project file, re-slice, and print.

Now I am noticing that even after saving the project file over top of the old one, if I then re-open the 2.3 project file there are random settings that are not coming back in the way they are saved. This is intermittent behavior and the settings are random (to me at least) so I don't have screenshots of them yet.

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
Screenshots attached of what settings are different in 2.3 when opening a project file saved in 2.2.
[Pictures.zip](https://github.com/prusa3d/PrusaSlicer/files/5875197/Pictures.zip)

rtyr, 2021-01-27:
I don't see anything wrong with the attached screenshots, it mostly reflects the changes we made in last configuration updates. The values with orange arrow and lock were changed in our system profiles over time, so this just indicates, that your saved value (which was previously used as default) is different from current value in the system profile. It is expected behavior.

Some examples from your screenshots:

Fan speed at layer - this setting did not exist in PS2.2.0, so we import value 0. Zero is different from default value.
Slow down if layer print time is below - Default value was previously 20s (when you saved your project), it was later changed to 15s for this filament.
[image]

Default z-lift was previously 0.6mm, it was later changed to 0.4mm. Saved value doesn't match with current system value, therefore the orange indication is expected.
[image]

Same here. The default start g-code and end g-code in system profile was changed, since you saved your project. Your saved values are different from current values in our system profiles = expected behavior.
[image]

pigskins65, 2021-01-27:
I understand, thank you.

closed 2021-01-27 by pigskins65