prusa3d/PrusaSlicer #5897: brim does not appear after slicing

wimjan1, 2021-01-26:
### Version
2.3.0+win64
build: PrusaSlicer-2.3.0+win64-202101111315

### Operating system type + version
Microsoft Windows 10 home
version: 10.0.18363

### 3D printer brand / version + firmware version (if known)
home built cartesian printer
Firmware: repetier version: unknown

### Behavior
skirt and brim

 do not show up in preview mode but do get printed.

#### Project File (.3MF) where problem occurs

[image]
[image]

rtyr, 2021-01-26:
Please provide your 3MF project file.

wimjan1, 2021-01-26:
[3mf brim issue.zip](https://github.com/prusa3d/PrusaSlicer/files/5875827/3mf.brim.issue.zip)
here you go.
thanks for the quick reply and good luck.

wimjan1, 2021-01-26:
> Please provide your 3MF project file.

uploaded the 3mf zip to the first post

foreachthing, 2021-01-26:
Probably doesn't help (much), but if I import your model into my settings all is fine:

[image]

If I open your 3mf directly .. same issue you have.
So, _my_ guess is, your profile does something odd ...

wimjan1, 2021-01-26:
I have a z-bias of -0.2mm maybe that is the cause. it does show the brim and skirt while slicing but they just disappear when the slicer is done.

wimjan1, 2021-01-26:
> Probably doesn't help (much), but if I import your model into my settings all is fine:
> 
> [image]
> 
> If I open your 3mf directly .. same issue you have.
> So, _my_ guess is, your profile does something odd ...

maybe it is because I have a z-bias of 0.2 mm and a first layer of the same height. so it might not show it. can't think of any other reason though.

foreachthing, 2021-01-26:
> maybe it is because I have a z-bias of -0.2 mm and a first layer of the same height. so it might not show it. can't think of any other reason though.

You can change your Z-Offset to -0.0199 and it will render. 😮

wimjan1, 2021-01-26:
> > maybe it is because I have a z-bias of -0.2 mm and a first layer of the same height. so it might not show it. can't think of any other reason though.
> 
> You can change your Z-Offset to -0.1995 and it will render. 😮

just tried it and it works with z bias= 0.1995. this is a brilliant intermediate solution but if there is a setting or something which can be switched on to also render the path below z= 0 mm that would be great. this can also be seen as a safety feature as you might want to know if your code is doing something below z= 0 mm when it shouldn't.

rtyr, 2021-01-29:
I understand what do you want, but on the other side, the preview is actually correct from the g-code point of view. You will get the same result in other slicers with such g-code.

Personally, I think the best option here is to adjust things on the printer side and not use the z-offset at all.

closed 2026-09-29 by prusaslicer-bot[bot]