prusa3d/PrusaSlicer #5900: Wipe Tower - Linear Advance GCode can be incompatible with machine.

kbrickman24, 2021-01-26:
### Version
2.3.0 RC2

### Operating system type + version
Windows 10 X64

### 3D printer brand / version + firmware version (if known)
Blackbox ToolChanger / Duet 2 / Firmware 2.05

### Behavior
This request is an extension of previously closed issue here: [](https://github.com/prusa3d/PrusaSlicer/issues/3965)

This issue was "resolved" and pushed to the main branch.

 """When starting wipe tower gcode, linear advance (pressure advance) is disabled. For Marlin fw this is done properly with M900 K0. However, for reprap, slicer ouptuts M572 D0 S0. This is valid only for tool (extruder) 0. When wiping extruder 1, command must be M572 D1 S0 and so on."""

The problem is that this solution assumes that each tool has it's own extruder motor. In the Case of Blackbox, a single extruder motor drives all tools. Duet reports hard errors on every tool change as the Gcode is attempting to turn off linear advance for an extruder that does not exist (I.E. M572 D4 E0) We will need a way to separate these types of machines.

#### Project File (.3MF) where problem occurs
[dualexttest.zip](https://github.com/prusa3d/PrusaSlicer/files/5876301/dualexttest.zip)

kbrickman24, 2021-02-02:
Correction - "(I.E. M572 D4 S0)"

Bi00, 2021-07-01:
I have the same problem with my Duet 3 firmware 3.3.0 and a Prusa like MMU.
Only 1 motor, 5 tools.

Kbreezy24, 2026-09-02:
https://github.com/Kbreezy24/blackbox/tree/master/Firmware%20%2B%20Slicer

It's been so long now I can't remember how we solved for this, but you can find the current Blackbox Firmware & Slicer files (Now Duet 3) on the github here!