prusa3d/PrusaSlicer #5901: Firmware Request! Ability to resume print when "SD card removed" - faulty hardware 🙁

rw3iss, 2021-01-26:
### Version
2.3.0+linux-x64

### Operating system type + version
Ubuntu 20.10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+

### Behavior
Feature Request:
Looks like my SD card reader has become faulty... constantly thinks the SD card is being removed... I will try and open it to adjust the pins a bit, but there should be some option to resume the print if the printer detects the SD card has (accidentally) been removed, similar to crash detection or power failure...

neophyl, 2021-01-26:
Wrong GitHub project. You need to post in the one for your printers firmware. I believe you have a mk3 so that would be https://github.com/prusa3d/Prusa-Firmware/issues.

rtyr, 2021-01-26:
As neophyl said. Closing.

closed 2021-01-26 by rtyr

rw3iss, 2021-01-26:
Ah shyt, thanks.