prusa3d/PrusaSlicer #5902: Bug report: Size modifiers do not listen to direct size inputs, only percentage scaling.

Incendium87, 2021-01-26:
### Version
2.3.0

### Operating system type + version
Win10

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
_Describe the problem:_
* When changing the size of a sub part by directly inputting a size in mm, the result is unpredictable. The part is not changed by the specified amount and will change the size in the other directions even if the scale of the part is unlocked. It almost looks like the part is being scaled by the percentages that are already listed above. 
* Its possible that the mm size that is inputted is being scaled by the original model scale. This may make sense from a hierarchy standpoint, but is not the behavior I would expect as a normal user.
* This only happens if the overall part has already been scaled by some amount

_Steps needed to reproduce the problem:_
* Scale overall model by some amount using the top item in the tree
* Try to change some sub part (support enforcer, or the original model for example) by a specific mm input
* Then the part changes in an unexpected way

_Expected Results:_
* part changes to specified size

_Actual Results:_
* part changes to an odd size that is not specified. Appears to act correctly if a scale factor percentage is inputted instead.
* _Screenshots from __*PrusaSlicer*__ preview are preferred:_
[image]
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Slicer Size Scaling bug report.zip](https://github.com/prusa3d/PrusaSlicer/files/5876682/Slicer.Size.Scaling.bug.report.zip)

mucek4, 2021-01-27:
Just find out that myself. It very annoying.

neophyl, 2021-01-27:
You are using World coordinates.  When you select the top level Bernie part, use the dropdown and change world co-ordinates to local coordinates.  The select you modifier and scale an axis.  Try that.

I may be wrong but its my understanding that with World all things are interlinked, so you scale one thing and it effect everything else.  When you select local it applies things just to itself.  I tried in in PS 2.3.0 on Windows 10 this morning and I could scale your modifiers in a single axis and not effect any other modifier or the Bernie.

Battlevaria, 2021-01-31:
Just hit the same trying to add a base to a figurine. Having a sub-part of a shape with a scale leads to very confusing results. Once I made the base a separate part, it stopped misbehaving. But then I couldn't lift the figurine onto the top of the base. I eventually got to where I wanted by calculating the scale percentages, switching the whole thing to "Local coordinates", then offsetting the figuring by the base height but adjusted by the scale factor. Once you scale the individual subparts first, the whole thing makes more sense. Not a lot of sense, but enough sense to get what I wanted.

togomedia, 2021-02-02:
I have experienced this as well. any change to absolute size if recalculated as a percentage when dimensions are unlocked. The calculation should be based on percentage OR absolute size, depending on which input is made.

Thank you for bringing this up. ~Mark

marcus7070, 2021-02-16:
I can recreate this with with generic shapes and modifiers as well, no need for an input file.

1. right click -> add shape -> cylinder
2. scale cylinder to x=100mm
3. right click on object -> add modifier -> cylinder
4. scale cylinder to x=90

result is x=270.47 when you expect 90.

Order of the steps is important, you can work around this by scaling the modifier before you scale the base object.

v2.3.0 on Linux (NixOS)

lual, 2021-04-30:
I can confirm this bug.

Thank you @marcus7070 for your workaround:
> Order of the steps is important, you can work around this by scaling the modifier before you scale the base object.

closed 2026-09-29 by prusaslicer-bot[bot]