prusa3d/PrusaSlicer #5903: Provide a placeholder "next_extruder_extruded_volume" to custom toolchange gcode

spiky2021, 2021-01-27:
### Version
2.3.0

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)
Anycubic I3 Mega S, Craftbot Flow IDEX XL
### 
Many users of multi extruder printers like to define their own purge/retract procedure during toolchange. Now the problem occurs, that PS alway adds a travel with retract in case an extruder wasn't used before and thus wasn't retract by a toolchange retract. Meaning without information about the actual extruded volumen of the next extruder a custom toolchange gcode can't decide wether the sclicer adds a travel retract or not.  To gain control in the custom gcode, one can use the workaround to bring all first extrusion to the bed (partially described in issue #5823) , but this is much less then user friendly.
Therefore I added an additional placeholder parser key to the set_extruder method of the GCode Class.
The toolchange gcode I use for a Craftbot Flow IDEX XL and the GCode.cpp files are attached.
[additional_placeholder_custom_gcode.zip](https://github.com/prusa3d/PrusaSlicer/files/5879521/additional_placeholder_custom_gcode.zip)
I know, that adding another placeholder interferes with the standard documentation of Slic3r placeholders.
But if PrusaSlicer not extends at this point, it will stay in the bounderies defined by a more or less historical Slic3r.
It is a small change, but intrinsically opens PS to a much wider range of newly developed printers and users, developers, etc..   
_Is this a new feature request?_ Yes

#### Project File (.3MF) where problem occurs

bubnikv, 2021-01-27:
Would you please provide a pull request through github? Or at least a diff?

spiky2021, 2021-01-28:
Hm, well my VS2019 is not linked to the repository so far, I just downloaded the sources. Seems like I have to do that, then to create a fork/branche and then I'm able  provide a pull request? Sorry, if I'm wrong, I'm completely new to git.

spiky2021, 2021-01-30:
Now provided as pull request combined with changes under #5823.

spiky2021, 2021-02-07:
Dear Vojtěch Bubník, 
I provided a pull request as you requested to me. But since that I havn't got any responce! Maybe I did something wrong, but then please inform me about your guidelines. Otherwise I have to percieve it as rude to request something and not to respond after it is done. I worked together with teams from universities like MIT, KIT, ETH, NSU, etc., as well as institutions like NASA, CERN, Academia Sinica, etc., as well as international companies and this was always commen sense. Therefore I expect  the same for any open source community, if it is really open minded. Maybe this is overreacted, but I feel a little offended, since i solved some issue placed about 4 years ago and nobody is interrested about that.

bubnikv, 2021-02-15:
Merged with 32db22b77c73dd1579da686b748267ce5ee1aeb5
Thanks for your contribution.

closed 2021-02-15 by bubnikv