prusa3d/PrusaSlicer #5904: Increased fragility of the product when using ironing

Tomikant, 2021-01-27:
### Version PrusaSlicer 2.3.0 RC1

### 3D printer Prusa MK3S

### Behavior
The increased fragility of the product was reflected in the use of ironing in my case. I printed products for which printing continued further from the layer level where ironing was used. These were vertical walls with a thickness of 0.6 to 3 mm. The walls were very easy to break off at the level of the layer, where ironing was used (see picture). I ruined several products, all-day printing. I had to reprint everything without using the ironing function.
I ask for your experience.

Profile: Plasty Mladeč PLA
Filament Plasty Mladeč PLA
Layer 0.2 mm 

[image]

BR Tomikant

rtyr, 2021-01-27:
It is likely caused by heat creep in the heatbreak as mentioned here https://help.prusa3d.com/en/article/ironing_177488.

fe60, 2021-02-12:
To prevent the forces shown in your picture to break the walls off, the connection between the bottom and the inner perimeter of the first layer above the bottom is pretty important. And according to your post, this seems to become weaker as soon as ironing is performed. According to the preview in PrusaSlicer, the surface ironed, it slightly larger than the visible surface, which could impact the adhesion between the layers **(just guessing)**.
Did you try to activate ironing using a modifier which is slightly smaller than the surface to be ironed? That way the layer below the first perimeter won't be touched by the ironing process.
If required you can design and load your own adapted modifier to match the shape of the surface to be ironed.

Tomikant, 2021-02-16:
Thank you for your contribution. I believe that the space to be ironed would have to be much larger than the exposed horizontal surface. It can be seen in the photo that the product contained three vertical walls of different thicknesses. All these walls broke very easily at the moment of removing the products from the printing pad. I did not further test the procedure with the modifier, I did not have time to try something and so I printed all the pieces of products without ironing. These are fully functional and error-free. I haven't made any more attempts yet. That is why I enclose the used g-codes. Assuming that this is a recurring phenomenon, I believe that there is a serious shortcoming in the slicer if it cannot properly treat those surfaces on which additional material will be applied. Does anyone have a similar experience to me? 
[image]
[2Klapka _ 01_0.2mm_PLA_MK3S_5h46m_Ironing_on.zip](https://github.com/prusa3d/PrusaSlicer/files/5988454/2Klapka._.01_0.2mm_PLA_MK3S_5h46m_Ironing_on.zip)

michael-kerbel, 2021-02-21:
Yes, your experience is not unique, see e.g.
https://github.com/prusa3d/PrusaSlicer/issues/4950
https://github.com/prusa3d/PrusaSlicer/issues/5113
https://github.com/prusa3d/PrusaSlicer/issues/5634

As @rtyr already noted, this happens due to heat creep. I did find a manual workaround, but unfortunately it is rather annoying to set up:

https://github.com/prusa3d/PrusaSlicer/issues/5113#issuecomment-747602704

stepikovo, 2022-03-23:
I'm closing, this is a duplicate of #5634

closed 2022-03-23 by stepikovo