prusa3d/PrusaSlicer #5905: First layer comes out on top

danny1231233, 2021-01-27:
Hi, I am using prusaslicer version 2.2.0 and whenever I try to slice certain models, the first layer gets removed and comes out on the utmost top not even connected to the rest of the part. As a reference view the image below which is a plane that I am trying to print but some sort of top layer comes out on top for some reason. I added an image before slicing and after.

[image]

[image]

rtyr, 2021-01-27:
Please provide your 3MF project file.

danny1231233, 2021-01-27:
@rtyr I am not sure how to do that but I can add it as an stl or f3d

danny1231233, 2021-01-27:
https://drive.google.com/file/d/1Vo-pEM_ivAHAa8Vwldm98hVdOZWm9TiA/view?usp=sharing      

Here's the link to download the .stl then click split to parts in prusaslicer when you open it.

rtyr, 2021-01-27:
> I am not sure how to do that

File -> Save project as....
[image]

You need to zip it first if you want to upload it directly to the GitHub. Or you can just use google drive again.

I think the issue is caused by your printer settings, so I need whole 3MF project or at least the config (File-Export-Export config).

danny1231233, 2021-01-28:
https://drive.google.com/file/d/1MDLnmSW-cdjF-Civ5hpmy6GY9WvcqkR6/view?usp=sharing

ok, here it is. and thanks again for the help.

rtyr, 2021-01-28:
It is caused by the custom g-code in your filament profile.

Remove this line, it will fix the issue.
[image]

danny1231233, 2021-01-28:
That worked. Thanks a lot for your help.

closed 2021-01-28 by rtyr