prusa3d/PrusaSlicer #5906: Print object is empty after rotation

hilleke-dirk, 2021-01-27:
### Version
Version 2.3.0+win64

### Operating system type + version
Win 10

### 3D printer brand / version + firmware version (if known)

### Behavior
After a rotation of 45 degree, PrusaSlicer reports "The print is empty. The model is not printable with current settings"
An outside rotated model is printed ok.
[seifenschale3_problem.zip](https://github.com/prusa3d/PrusaSlicer/files/5881385/seifenschale3_problem.zip)

neophyl, 2021-01-27:
Your attached 3mf sliced fine for me.  PS did report 24 auto repaired errors though.  The attached stl has 85 intersecting faces and 12 zero faces and 6 zero edges according to Blenders 3d print tool.

Sometime saving as a 3mf and then loading will 'fix' the model whereas just loading in the stl and slicing doesnt.  Not sure why.

hilleke-dirk, 2021-01-27:
I put the stl's in the zip too. (I hope)
_scheibe.stl works
_problem.stl rotated by 45° do not work

neophyl, 2021-01-27:
Yes you did.  However as I pointed out the stl file you included has errors.  I have attached a zip of a fixed version.  It slices in all the angles I tried.  I used the place to face tool to try multiple orientations too, although after placing to face I had to rotate to fit it to your bed size.  
[seifenschale3_problem_neo.zip](https://github.com/prusa3d/PrusaSlicer/files/5881811/seifenschale3_problem_neo.zip)

closed 2021-01-28 by hilleke-dirk