prusa3d/PrusaSlicer #5907: Filament used for multiple extruders

Alphaprot, 2021-01-27:
The `used_filament` placeholder command is a sum of all filaments used in all extruders. Is there a way to implement a more granular way to query the filament per extruder or per material type?

Example: `used_filament_0` for extruder/tool 0 and `used_filament_1` for extruder/tool 1 or `used_filament_filament_type_x` for all filament of a certain material (e.g. if doing multi-material prints with pauses to swap the filament). I myself would be more interested in a per-extruder approach.

In case you are wondering why I am proposing this feature - I own an IDEX printer (FlashForge Creator 3) and are currently tinkering to get it working with PrusaSlicer, mainly with custom g-code and a script based on gx.py (https://github.com/ronoaldo/FlashforgeFinderIntegration) , which I'm extending for more functionality (namely replacing unsupported Marlin commands with their corresponding FlashForge equivalent) .

spiky2021, 2021-01-28:
Hi, please check #5903, it concerns the same topic. What I don't understand is why you need a post processing. If one uses a custom gcode like mine, attached to above topic, there are no further printer specific gcode statements (M..) written? Well if you  use variable fan speeds, yes you need it. As I remember PS just writes "106 S..." without the corresponding Tool. Beside a preheating algorithm for the next tool (hopefully without postprocessing), this is another topic I like to work on in future. Something like, if "num extuders >0" write "M106 S.. T.." else "M106 S..". My further interest is an additional custom gcode parser  allowing to exchange printer specific gcode tags by certain placeholders, like pause, etc..

Alphaprot, 2021-01-28:
Custom g-code post processing in my case means replacing/changing unsupported commands and adding a binary header to the new .gx file (instead of .gcode).

Based on my understanding, the existence of custom g-code for `Action_A` suppresses PrusaSlicer-generated g-code for `Action_A` only if a certain command is recognised; for tool changing, this command is `T{tool_index]`.

FlashForge printers use their own flavour, which supports some Marlin commands, while others are proprietary. E.g. a tool change happens with `T108 T[tool_index]` (FlashForge) and not with `T[tool_index]` (Marlin). And that's the reason I cannot simply use the custom tool-change g-code section to make said post processing script obsolete - the custom code requires a command I cannot use and have to replace afterwards.

spiky2021, 2021-01-28:
I see, didn't know that some firmwares have this. Actually bad style, since none M.. gcode is a quite old standard. Anyway you are right, the custom gcode is parsed for certain prefixes, you can find in method "toolchange_prefix" under source link https://github.com/prusa3d/PrusaSlicer/blob/master/src/libslic3r/GCodeWriter.cpp . Your flavor is not supported so far.

spiky2021, 2021-01-28:
I was really wondering, that floashforge uses something like T108. By search I only found, M108 for flashforge, which is sailfish flavor. Means, it may help, if you try  this flavor in combination with a proper toolchange gcode?!

Alphaprot, 2021-01-28:
Ouch, my bad… I misspelled that piece of code – you are right, it should be `M108` instead of `T108`. Frankly, I did not know that this is Sailfish-flavor (unfortunately, Sailfish is poorly documented; there is literally no g-code reference available). As Sailfish seems to be a thing from the past (the official repo was last edited 4 yrs ago), I will probably stick to Marlin (this is actually recommended by FlashForge support – idk why) and post-process-fix those little differences. But yeah, you are right; that won’t stop me from implementing some better tool-changing behaviour (like the one you did in your custom g-code).

h8nor, 2022-11-01:
[Already mentioned](8894) the proprietary printers S3D. The original gcode has a lot of obscure commands.
Each new layer is immediately preceded by command `M103`, but it can be preceded by command `M108 S<float>` for different types of objects _SKIRT, WALL-INNER, WALL-OUTER, FILL, SKIN_.
End movement with `M101` + `G1 X-<float> Y-<float> Z<float> F<float>`

closed 2026-09-29 by prusaslicer-bot[bot]