prusa3d/PrusaSlicer #5910: [BUG] Slicing hangs when Infill Percentage set very low

fiveangle, 2021-01-27:
Slicing hangs when Infill Percentage set very low

2.3.0+win64
2.3.0+macOS

To reproduce:
-Open the attached project file
-Select latest "0.3mm DRAFT" printing profile
-Select latest "Generic PLA" filament profile
-Select latest "Original Prusa i3 MK3" printer profile
-Change "Infill density:" to "0.0001"
-Click "Slice"
-Observe CPU spike to 100%
-Observe application go non-responsive
-Observe status info bar show "Preparing infill..."
-Observe application hangs in this state indefinitely
-Observe application will not quit from within OS (both Windows 10 and macOS)

[TEST_Left_Bracket_inside.zip](https://github.com/prusa3d/PrusaSlicer/files/5883580/TEST_Left_Bracket_inside.zip)

bubnikv, 2021-02-05:
Fixed with aa4c018cd5857814ed9ac0cd93d526f3f172d4ab

    Fix of Slicing hangs when Infill Percentage set very low #5910
    When passing a PrintRegionConfig to the slicing back end, zero an infill
    density if it is already nearly zero. This avoids an ugly division by
    zero in the infill code and it also makes clear to the back end that
    we don't want an infill.

Closing this one.

Now I expect that you were using that very low infill density to modify behavior of some other algorithm, to work around some other bug, right? Which one was it?

closed 2021-02-05 by bubnikv

fiveangle, 2021-02-05:
> Now I expect that you were using that very low infill density to modify behavior of some other algorithm, to work around some other bug, right? Which one was it?

Yeah, it was the inability to use the _Solid Layer Every XX_ feature which I filed #5909 against.

I found another user opened a similar issue, but proposed it as an enhancement request, but since solid layers are always 100% infill, a user would never expect setting infill at 0% would affect the feature.

Thanks for circling back on this one Vojtěch ! ❤️