prusa3d/PrusaSlicer #5912: PrusaSlicer 2.3.0 did not create supports in a special case

hackyber, 2021-01-28:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_2.3.0

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_Windows 10
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) Prusa MK3+
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_Prusa MK3+

### Behavior
* _Describe the problem_
If I try to slice the enclosed STL-file inthe origin orientation with supports under the hollow area it will not do the job. If I turn the
part onto another side ist works fine. For my project it is ok now. But
[image]
 I believe this is an error.
[Schraubstockbacken.zip](https://github.com/prusa3d/PrusaSlicer/files/5887290/Schraubstockbacken.zip)

* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

22chrs, 2021-01-28:
same problem here.

22chrs, 2021-01-28:
testet in Mac and Windows operating system. Same issue on both.

rtyr, 2021-01-28:
Correct behavior. Disable "Don't support bridges" option, if you want supports there.

[image]

rtyr, 2021-01-28:
Can we close the issue?

hackyber, 2021-01-28:
Yes, now it works. But anyway it is a suprising behavir if I switch support to "everywhere (Überall)"

Best regards Bernd

[image]

rtyr, 2021-01-28:
It is everywhere like "not only from the build plate".

OK, closing.

closed 2021-01-28 by rtyr