prusa3d/PrusaSlicer #5916: What is "MultiAsSingle" Mode for G-Code ticks used in the Double Slider view? And how to add *Filament* changes like Color changes to layers?

scottmudge, 2021-01-28:
### Version
v2.3.0

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S+ (3.9.3)

### Behavior
When using the double-slider in the sliced object view, there are 3 options -- Change Color (M600), Add Pause (M601), and Custom GCode.

Unfortunately the "Change Color" option only opens a color picker. Which doesn't really add any benefit, except for visualization.

I would like the ability to select a new **filament profile** (of the same material type) when adding a Color Change. I have a few different brands of PETG which work well at different temperatures and cooling parameters, so inserting an `M104` command (or fan speed changes) after the `M600` filament change command would be ideal.

I was looking into adding this feature myself, but as I was looking at the source code, I noticed that there is also a `ToolChange` event/tick type, which is only enabled for Multi-extruder setups.

However, there is also a "MultiAsSingle" extruder mode:

```
MultiAsSingle,    // Multiple extruder printer preset is selected, but 
                      // this mode works just for Single extruder print 
                      // (The same extruder is assigned to all ModelObjects and ModelVolumes).
```

But it is unclear what this is, or how it can be activated (if at all).

Is it possible to use this mode with a ToolChange type tick on a single extruder system, to change the active filament profile to a different filament profile at a certain tick, rather than just using a simple M600 command?

rtyr, 2021-01-28:
> I would like the ability to select a new filament profile (of the same material type) when adding a Color Change. 

Already requested here https://github.com/prusa3d/PrusaSlicer/issues/2002. 

Basic things can be done using "add custom g-code" option (you can use more commands at once, like M600, M104, M106), it should be enough in most cases.

> But it is unclear what this is.

https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha4, Colorprint improvements, point 2.

> Is it possible to use this mode with a ToolChange type tick on a single extruder system, to change the active filament profile to a different filament profile at a certain tick, rather than just using a simple M600 command?

No, it is used for printer with multiple extruders. You can enable MMU2 profile to see how it works.

scottmudge, 2021-01-28:
I see, thank you.

Yes I would like to avoid writing custom g-code. I may work on this feature on my own time then, depending on the scope of the work. Reproducing the filament profile/preset combo box in a separate selection window looks like it will be quite involved, so it might just be a long term wish.

Appreciate the links.

closed 2021-01-28 by scottmudge