prusa3d/PrusaSlicer #5918: Z-offset critical bug

ailireruf, 2021-01-29:
### Version
Version 2.3.0+win64

### Operating system type + version
Windows 10 64 bit (17763)

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 V2 with upgraded bed springs and BLTouch 2.0 self leveling. Firmware 1.1.1

### Behavior
I've set my Z-offset to -2.4mm ( this is correct height for my set up) in the printer settings. 
Issue 1: After slicing the object is shown to be partially beneath the print bed. You can clearly see it on screen, and Octoprint gives warning that the object does not fit. 
Issue 2: When printing the first couple of layers. everything is fine. However, once the print get higher than about 2.4 mm ( I specifically measured that), the print head start making large blobs and then goes way too high. I suspect by exactly 2.4 mm. 
As a result prints fail. 
This is not an issue of parts or printer. I tried several different parts with the same result. If I create g-code using Slic3r on exact same part with all identical settings, including the z-offset, the prints come out fine. 
Please help! Thank you!
[z-offset bug 1-28-21.zip](https://github.com/prusa3d/PrusaSlicer/files/5891018/z-offset.bug.1-28-21.zip)

rtyr, 2021-01-29:
PrusaSlicer is just doing what you asked and then showing the result in the preview. You will get similar preview to such gcode in other slicers. That is also the reason, why the Octoprint doesn't like this. You are printing outside of the build plate (from the g-code point of the view).
You should fix your issues on the printer side, in the start g-code or in the firmware. 

[image]

S3D
[image]

Cura
[image]

CraftWare
[image]

ailireruf, 2021-01-29:
I understand the part being under the bed. It more of a inconvenience. 
However it does not explain why the first couple of layers are printer correctly and after that the print head is moved way too high. 
Its like once it gets above the z-offset value, it defaults to normal height and raises the print head by that value. This is just speculation though. 
Again, this problem does not happen when sliced with Slic3r or other software. Every printed layer is consistently offset by -2.4 mm. Here, when the print get to about 2.4 mm height, the z-offset defaults to 0. Or so it appears.
If there is a gcode command to fix this, program permanent z-offset, which command is it? Thanks

rtyr, 2021-01-29:
> However it does not explain why the first couple of layers are printer correctly and after that the print head is moved way too high.

Maybe it would be better if you tell us the line in the g-code where do you think is some issue. What I can see from your gcode:
```
Layer 10 G1 Z-0.600
Layer 11 G1 Z-0.450
Layer 12 G1 Z-0.300
Layer 13 G1 Z-0.150
Layer 14 G1 Z-0.000
Layer 15 G1 Z0.150
Layer 16 G1 Z0.300
Layer 17 G1 Z0.450
```

Maybe it is just about your MIN_POS_Z value in your firmware.

pmjdebruijn, 2021-02-12:
@ailireruf this may be slightly besides the point, but usually you'd setup the Z-offset in your printer firmware, as opposed to in the Slicer.

rtyr, 2021-03-04:
No response, the g-code looks ok. Closing.

closed 2021-03-04 by rtyr