prusa3d/PrusaSlicer #5919: G10 command missing tool number for RepRapFirmware

grtyvr, 2021-01-29:
### Version
Version 2.3.0+linux-x64

### Operating system type + version
Ubuntu 20.4

### 3D printer brand / version + firmware version (if known)
Duet RepRapFirmware 3.2

### Behavior
Generated gcode in section above the TYPE:Custom tries to set tool temperature but does not specify a tool.

Generates

G10 S<temp>

But should generate

G10 P<xxx> S<temp>

where <xxx> is the tool number.  There does not seem to be a way to specify what tool to use.

rtyr, 2021-01-29:
Using this in the start g-code should work `G10 P[initial_tool] S[first_layer_temperature]`

grtyvr, 2021-01-29:
That is what I have done for a workaround.

rtyr, 2021-01-29:
So the problem is you expect slicer to add G10 P0 Sxxx? RepRapFirmware will not use P0 automatically if not specified?

rtyr, 2021-01-29:
Or maybe you are talking about this? https://github.com/prusa3d/PrusaSlicer/issues/1796

grtyvr, 2021-01-29:
It might be related to that.  I have not tried using Slic3r with my RepRapPro Tripple so I can't say for sure.  What I have observed is that if the printer is configured with a single tool, then the code generated prior to the inclusion of the start g-code includes the command G10 S[first_layer_temperature].  That throws an error in the RepRapFirmware about no tool selected.  So either there needs to be a T0 prior to the G10, or the G10 needs to be G10 P[initial_tool] S[first_layer_temperature]

grtyvr, 2021-01-29:
And to answer the previous question directly: yes, RRF will not use P0 automatically.  You need to specify the tool.

bubnikv, 2021-12-20:
Fixed with #7536
thanks @rsobik for contribution.

closed 2021-12-20 by bubnikv