prusa3d/PrusaSlicer #5921: Buggy custom G-Code / 0 Temperature - Printed model once fine, then printed same model again and temp always set to 0

rw3iss, 2021-01-29:
### Version
_Version of PrusaSlicer used goes here_
2.3.0 linux 64-bit

### Operating system type + version
Ubuntu 20.10

### 3D printer brand / version + firmware version (if known)
MK3S + latest firmware 3.9.3

### Behavior ###
First, I upgraded to latest slicer and firmware. I set my filament/print settings as usual... print temperature 215 for both initial and all layers. 
When the print started, it went right to 170F temperature, oddly, but it didn't change.
I then looked all throughout the settings, but nothing. I then looked at the custom G-CODE and noticed there was a code to prime the printer to 170F. I removed these (two) lines). I then exported and sent to printer, and it printed fine.
The strange part is... I then printed the same model, from the same SD card (without removing it), and the second print set the temperature to 0, and wouldn't change. Print bed temperature is fine. 
I went back into PrusaSlicer, checked settings, re-exported, and same thing... 
I then looked at custom G-CODE, and all the custom G-CODE has changed! It's totally different from what I saw the first time. Explain that to me.
Anyway, here is the custom G-CODE node. It looks this way for all filaments.
Now everytime I print the temperature is always 0.
What gives?
I tried other slicers/exports - they work normally. So seems a bug with the custom G-Code getting replaced... for some reason...

#### Project File (.3MF) where problem occurs
[image]

Also the uploader here doesn't support .3mf filetypes, fyi...
Here is the .3mf file:
https://drive.google.com/file/d/1xFHELYaKURokU6Yq_66bKyBsOzCigNfq/view?usp=sharing

rw3iss, 2021-01-29:
I think I see what happened...
I cleared out all prusa files and config, everything, and redownloaded/installed the program, re-initialized. I selected MK3S, my printer version, but then when I went to Expert mode > Printer Settings, it defaults to the MINI settings! I changed it to the MK3S settings and all the G-Code has changed. 
Anyway, I changed it to MK3S, and it seems to have cleared up the problems.
I don't know if the defaulting-to-MINI thing is actually a bug... strange that it was still there even after deleting all Prusa files from machine, and re-configuring (did a system search for 'prusa'). Anyway...

closed 2021-01-29 by rw3iss

rtyr, 2021-01-29:
I believe you simply chose incorrect printer profile (Prusa MINI instead MK3S).

rtyr, 2021-01-29:
ok

rw3iss, 2021-01-29:
Weird thing is I just started the print... it fired up temperatures, but it's not doing anything else... just sitting at 215 and nothing. I didn't modify the G-Code this time... will play around with it.

rw3iss, 2021-01-29:
OK disregard I guess - reset machine and re-exported and seems to be back to normal.