prusa3d/PrusaSlicer #5922: Cant make support work

batata004, 2021-01-29:
Hi,

Every time I enable support "Everywhere" or any other option of support I get the error message `support_material_extrusion_width=0.26 mm is too low to be printable at a layer height 0.3 mm`. If I change the `Default extrusion width` from `0` to `0.3` then I get another error message `extrusion_width=0.3 mm is too low to be printable at a layer height 0.3 mm`. If I disable supports all errors go away. So what should I do? 

My `PrusaSlicer-config.ini` is below:

```
avoid_crossing_perimeters = 0
avoid_crossing_perimeters_max_detour = 0
bed_custom_model = 
bed_custom_texture = 
bed_shape = 0x0,180x0,180x160,0x160
bed_temperature = 80
before_layer_gcode = 
between_objects_gcode = 
bottom_fill_pattern = monotonic
bottom_solid_layers = 2
bottom_solid_min_thickness = 0
bridge_acceleration = 0
bridge_angle = 0
bridge_fan_speed = 100
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
clip_multipart_objects = 1
color_change_gcode = M600
colorprint_heights = 
complete_objects = 0
cooling = 1
cooling_tube_length = 5
cooling_tube_retraction = 91.5
default_acceleration = 0
default_filament_profile = 
default_print_profile = 
deretract_speed = 0
disable_fan_first_layers = 3
dont_support_bridges = 1
draft_shield = 0
duplicate_distance = 6
elefant_foot_compensation = 0.2
end_filament_gcode = ""
end_gcode = G91 ; Altera o sistema de coordenadas para "relativo" para que o comando abaixo funcione de forma relativa.\nG1 Z5 F2000 ; Move o hotend um pouco para cima para que o recolhimento de filamento nao deixe marca do nozzle na peca.\nG90; altera o sistema de coordenadas de volta para "absoluto" que e o padrao da impressora.\nG1 E-6 F50 ; Recolhe 6 milimetros de filamento para evitar o pingamento de filamento enquanto o hotend esfria e enquanto ele esquenta na proxima impressao. Tambem serve para impedir que na proxima impressao haja filamento "frio" na ponta do hotend impedindo que a sonda se aproxime da heated bed o que pode inclusive danificar a impressora.\nM104 S0 ; Desliga hotend e nao aguarda ate chegar nessa temperatura.\nM140 S0 ; Desliga a heated bed e nao aguarda ate chegar nessa temperatura.\nG28 X0 ; Realiza o homing do eixo X.\nM84 ; Desliga todos os motores.
ensure_vertical_shell_thickness = 0
external_perimeter_extrusion_width = 0
external_perimeter_speed = 60
external_perimeters_first = 0
extra_loading_move = -2
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extruder_colour = ""
extruder_offset = 0x0
extrusion_axis = E
extrusion_multiplier = 1
extrusion_width = 0
fan_always_on = 0
fan_below_layer_time = 60
filament_colour = #29B2B2
filament_cooling_final_speed = 3.4
filament_cooling_initial_speed = 2.2
filament_cooling_moves = 4
filament_cost = 0
filament_density = 0
filament_deretract_speed = nil
filament_diameter = 1.75
filament_load_time = 0
filament_loading_speed = 28
filament_loading_speed_start = 3
filament_max_volumetric_speed = 0
filament_minimal_purge_on_wipe_tower = 15
[... 166 lines cut ...]
top_infill_extrusion_width = 0
top_solid_infill_speed = 60
top_solid_layers = 2
top_solid_min_thickness = 0
travel_speed = 130
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
variable_layer_height = 1
wipe = 0
wipe_into_infill = 0
wipe_into_objects = 0
wipe_tower = 0
wipe_tower_bridging = 10
wipe_tower_no_sparse_layers = 0
wipe_tower_rotation_angle = 0
wipe_tower_width = 60
wipe_tower_x = 180
wipe_tower_y = 140
wiping_volumes_extruders = 70,70
wiping_volumes_matrix = 0
xy_size_compensation = 0
z_offset = 0

```

rtyr, 2021-01-29:
You are trying to print 0.26mm wide extrusion together with 0.3mm layer height, which is outside of range for "sane" values. 0.31 will let you slice it, altough I don't think it is a good idea to print 0.3mm layers with 0.3mm nozzle.

batata004, 2021-01-29:
With my 0.3mm nozzle which is the layer height you would recommend me?

rtyr, 2021-01-29:
In general, don't go over 75-80% of the nozzle diameter. If you are new to the 3d printing, I wouldn't reccommend to go over 65% of the nozzle diameter.

I am closing this (no issue with PrusaSlicer).

closed 2021-01-29 by rtyr

batata004, 2021-01-29:
> In general, don't go over 75-80% of the nozzle diameter. If you are new to the 3d printing, I wouldn't reccommend to go over 65% of the nozzle diameter.
> 
> I am closing this (no issue with PrusaSlicer).

Thank you! I just realized 1 just have one nozzle, I thought I had another. So I will have to stick with 0.3mm nozzle BUT I still need the 0.3mm layer height (because of precision of holes of my objects are already calibrated). But I also need use support material, so how can I skip this "sanity check" that PrusaSlicer is complaining and preventing me from printing with support? Using S3D I could easily enable support with my 0.3mm nozzle and 0.3mm layer height with absolutely no problem.

So I ask you my friend, is there a way to disable this error message from Prusaslicer? Or is there any number I can change in the software (such changing from 0.3mm somewhere to 0.3001mm somewhere) to allow me to use supports?

rtyr, 2021-01-29:
Change the extrusion width for the support material to at least 0.31mm.

batata004, 2021-01-29:
@rtyr my friend I did exactly what you said, but I am not very smart and I am still getting error. See below, I changed the extrusion width as you told:

[image]

But after doing that and enabling support "Everywhere" I then see this error message:

[image]

What am I doing wrong?

batata004, 2021-01-29:
I am an idiot! I was changing the extrusion in the wrong place. Thank you so much my friend, sorry for taking your time!!!