prusa3d/PrusaSlicer #5923: Bug when using Tab to focus another input

batata004, 2021-01-29:
Hi,

Everytime I am focused on an input and I press TAB or SHIFT+TAB it works fine, I mean, the focus gets moved to the next/previous element and all of its content gets selected so I can start typing right away the new value. See print below, I had my focus in the ROTATE-X and when I pressed TAB the focus moved to the input ROTATE-Y and its content got all selected so it's pretty easy to me to change the value.

[image]

IMO this looks the "default" behaviour on PrusaSlicer and I like it. However in some other places this does not happen. Take a look at the print below:

[image]

My focus was at BED-FIRST-LAYER input box. When I pressed TAB the focus jumped to the next input element (that's correct) BUT instead of selecting the entire content of the input so I can start typing right away, it does not select the content and even worse, it puts the cursor at the middle of the `40` like `4|0`. In the print below you can understand this. 

If I was not clear enough, please tell me that I can provide more info.

fe60, 2021-01-29:
Not selecting the content of a field when entering it with (Shift+)Tab seems to be the default behavior for all input fields with the up/down buttons next to the field. There you get the selection/cursor position you have had focusing the field the last time or the cursor is in front of the content for the first time focus.

batata004, 2021-01-29:
For me, when using TAB (like in the first print) it selects the content of the input after giving it focus. It makes much more sense doing so because when I focus to an input, I certainly want to change its value. Makes no sense bringing focus to the element when using TAB (I am not talking about click because when you click you sometimes click between digits and want the cursor to be exactly where you clicked) and not selecting its content.

Also, what makes absolutely no sense is the behaviour I talked about on the second print because it makes no sense the input gets focused and the caret happens to be at a random position (does not matter if the input "saved" the last caret position when I left that input 1 week ago)... Anyway, from the usability and even accessibility, this is buggy and really bad. Just take a look at S3D and you willl see a great usability software but with terrible updates and terrible slicing seetings.

fe60, 2021-02-10:
@batata004 In the initial post you mentioned "some other places"  where you don't get the expected selection. I just wanted to make this a bit more specific. And you can observe this odd selection behavior for all input fields with up/down buttons.

batata004, 2021-02-10:
@fe60 the other places I dont get this buggy behaviour is like in the image below:

[image]

In this image, the behaviour is correct and expected when you tab over the inputs. However the behaviour is completely buggy/nonsense on the inputs of the imagem below:

[image]

I never ever found a software, app or even website that "saves" the caret position on each input when the user blurred the input just to use this caret position later when the user focus the input. Makes completely no sense. In a software like PS, people want to make fast changes on input values (mainly when they are numbers) so it makes a lot of sense SELECTING the entire content of the input when it gets focused and let the user set a new content (typing right away) or changing the content (using arrows to place the cursor where he wants).

YuSanka, 2021-02-19:
Thanks for your report. But it is not a PrusaSlicer bug but wxWidgets's. 
This Windows specific issue was fixed in upsteam wxWidgets with https://github.com/wxWidgets/wxWidgets/pull/1627. 
This bug will disappear from the PrusaSlicer when we update our dependencies.

closed 2021-02-19 by YuSanka