prusa3d/PrusaSlicer #5924: make test fails on a single test: "Scenario: PrintGCode basic functionality"

jensseidel, 2021-01-29:
### Version
git master d9448c9f2ad1e3ed869906ae5aac91596790308f

### Operating system type + version
MacOS 11.1

### Behavior
I fail to both compile and (after local fix) to pass the tests.

To fix the compilation problem I have to apply
```
diff --git a/src/slic3r/GUI/GUI_App.hpp b/src/slic3r/GUI/GUI_App.hpp
index 132776dcb..0038b41b2 100644
--- a/src/slic3r/GUI/GUI_App.hpp
+++ b/src/slic3r/GUI/GUI_App.hpp
@@ -221,7 +221,7 @@ public:
     virtual bool OnExceptionInMainLoop() override;
 
 #ifdef __APPLE__
-    void            OSXStoreOpenFiles(const wxArrayString &files) override;
+    void            OSXStoreOpenFiles(const wxArrayString &files);
     // wxWidgets override to get an event on open files.
     void            MacOpenFiles(const wxArrayString &fileNames) override;
 #endif /* __APPLE */
```

OSXStoreOpenFiles is a wxWidgets function defined at http://trac.wxwidgets.org/browser/git-wxWidgets/include/wx/osx/app.h and was never virtual. I am confused how to code compiles for you ...

After fixing this small issues all tests are fulfilled except:
```
% build.debug/tests/fff_print/fff_print_tests "Scenario: PrintGCode basic functionality" -v normal
Filters: Scenario: PrintGCode basic functionality
Testing Scenario: PrintGCode basic functionality
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Some text output is generated.
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Exported text contains slic3r version
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Exported text contains extrusion statistics.
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Exported text does not contain cooling markers (they were consumed)
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: GCode preamble is emitted.
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Config options emitted for print config, default region config, default object config
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Infill is emitted.
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Perimeters are emitted.
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: Skirt is emitted.
    Given: A default configuration and a print test object
     When: the output is executed with no support material
     Then: final Z height is 20mm
    Given: A default configuration and a print test object
     When: output is executed with complete objects and two differently-sized meshes
Assertion failed: (layer_id >= m_layer_id), function get_last_seam, file src/libslic3r/GCode/SeamPlacer.cpp, line 789.
```

The debugger reveals that m_layer_id == 99 and layer_id == 0. Is this test failing for you as well or can I expect that the master is always fine? Don't want to spend time on known issues ...

bubnikv, 2021-09-22:
Is this issue still relevant?

hexane360, 2022-12-30:
I can confirm this issue still occurs on git master (7eb2113).

I'm using wxWidgets v3.0.5.1 from MacPorts, on macOS 13.1.

closed 2026-09-29 by prusaslicer-bot[bot]