prusa3d/PrusaSlicer #5927: Feature Request: varying amount of perimeters on one part

Manni9000, 2021-01-29:
### Version
PrusaSlicer 2.3.0

### Operating system type + version

### 3D printer brand / version + firmware version (if known)

### Behavior
I was printing some corner brackets with holes for screws and needed solid material around the holes for strength. Of course I didnt want to print the whole brackets with 7 perimeters (which I needed around the holes), so I had to use the modifiers to change the amount of perimeters in the areas needed. After all this is quite a struggle. ( you have to  place a modifier for each hole and calculate the diameter exactly to match a correct amount of perimeters multiplied the extrution width. Also the perimeter at the edges of the modfier are seen as outer walls which is quite unnecessary.  You can see an example in the picture below.
While doing that I had the idea of a feature which lets you select walls of your stl and change the amount of perimeters for that surface similar to the feature which lets you select a surface and place it onto the print bed.
This would be a nice feature for structural parts to be a little more conveniet to print without just using the amount of perimeters needed at the weakest spot for the whole part.
If a feature similar to this already exists, please tell me where to find it.

_Is this a new feature request?_
yes

#### Project File (.3MF) where problem occurs

[image]

MonoTovarisj, 2021-06-01:
I agree changing only the inner perimeters in holes is a nightmare, the problem with the current modifiers is that they split the part in two and the slicer sees them as several objects if you have a more complex part that can be really dangerous.

Selecting a face/wall of an object and use as a modifier to change amounts of perimeters would be much better. Also would be nice to be able to use that wall modifier to up the part cooling fan for walls with excessive overhangs.

I have this current problem where I would like to add more perimeters on the inside as the part needs to hold preassure, however this part is only a cut out of a much larger piece that I cannot have more than 3 perimeters on, as it creates other issues.  Here I have tried to add a modifier, but as the slicer split the part in two, the places marked with red arrows won't get fused in the same way as if it was considered as one object.

[image]

Without modifier:

[image]

dartrax, 2022-03-15:
See #3846

closed 2026-09-29 by prusaslicer-bot[bot]