prusa3d/PrusaSlicer #5932: M601 Rendering issue in Layer view

INSOMNIAC84, 2021-01-30:
### Version
2.3.0+win64-202101111315

### Operating system type + version
Windows 10 (build 19042), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S+ (likely unlrelated)

### Behavior
Issue with M601 rendering in Layer view, when combined with height range modifier.
The Paused layer gets shifted up by 0.9mm (9 Layers in my case) which caused heavy confusion. The modifier was intended to add ironing to selected layers. But it also happens with a modifier not actually modifiying any settings.
The generated G-Code is correct and the print works as intended.

* _Steps needed to reproduce the problem_
- Load Model
- add height range modifier (i.e. 2 - 4,7mm)
- Slice
- Add M601 within the modifier (i.e. 4.3mm)

* _Expected Results_
M601 Layer should be colorized in the correct layer

* _Actual Results_
M601 layer appears shifted to higher/later layers (5.2 in my case)

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_
  
[image]

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[M601-gets-shifted.zip](https://github.com/prusa3d/PrusaSlicer/files/5897764/M601-gets-shifted.zip)

INSOMNIAC84, 2021-01-30:
Did some more testing:
Also happens when M601 is above or below modifier
Also happens without any modifier at all!

Prior to re-slicing after pause addition, the correct layer is rendered in grey.

rtyr, 2021-02-02:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/5780. We are aware of this issue, it is caused by the pause print G-code being inserted after Z-hop. We will fix it in the upcoming release.

Closing.

closed 2021-02-02 by rtyr