prusa3d/PrusaSlicer #5936: source code build fail on Ubuntu 18

sethu24, 2021-01-30:
### Version
PrusaSlicer 2.3.0
### Operating system type + version
Ubuntu 18.04.5 LTS ,intal 64 bit
Boost 1.65.1,wxWidgets3.0
### Behavior

../../src/libslic3r/liblibslic3r.a(Fill3DHoneycomb.cpp.o): In function `_GLOBAL__sub_I_Fill3DHoneycomb.cpp':
Fill3DHoneycomb.cpp:(.text.startup+0x3c): undefined reference to `boost::system::generic_category()'
Fill3DHoneycomb.cpp:(.text.startup+0x41): undefined reference to `boost::system::generic_category()'
Fill3DHoneycomb.cpp:(.text.startup+0x46): undefined reference to `boost::system::system_category()'
collect2: error: ld returned 1 exit status
tests/sla_print/CMakeFiles/sla_print_tests.dir/build.make:195: recipe for target 'tests/sla_print/sla_print_tests' failed
make[2]: *** [tests/sla_print/sla_print_tests] Error 1
CMakeFiles/Makefile2:1604: recipe for target 'tests/sla_print/CMakeFiles/sla_print_tests.dir/all' failed
make[1]: *** [tests/sla_print/CMakeFiles/sla_print_tests.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....

Building PrusaSlicer is failed in ubuntu , any one can help me to resolve this issues ?

combolek, 2021-02-03:
Not sure if this is the only problem but PrusaSlicer requires at least Boost 1.66 (see PR #5508).

lukasmatena, 2023-02-09:
Unable to reproduce, stale. Closing.

closed 2023-02-09 by lukasmatena